NimmLor/esp8266-fastled-iot-webserver #140: Apple Homekit Support or Hue

LordMM74, 2021-01-24:
Hello,
is there anyway to control it using Apple Homekit or using a Hue Bridge?

Regards

Jürgen

bb-Ricardo, 2021-02-01:
Hi Jürgen,

yes, there would be a way but a painful one.

I tried to integrate it but it does not work as reliable as expected. Only now and then the device is visible on the network due to missing MDNS announcement.

**You can try it yourself:**
* add library https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266 to your Arduino libs
* check out commit 32d1ad6ce47c6959508020d40f2769bb98e46296
* patch code with appended patch here
* compile and install

**Issues:**
* uses first 1407 Bytes of EEPROM data. This would erase all current config after upgrading to this code and config would need to set be again
* If library is not installed compiling fails because `homekit_led_accessory.c` gets compiled regardless if `ENABLE_HOMEKIT_SUPPORT` is set or not.
* it uses C macros extensively which makes it difficult to define HomeKit characteristics inside the .ino file.
* Pairing takes 10 to 15 seconds where LEDs don't respond
* On startup the mDNS announcement is very unreliable, the device shows up as offline during most starts
* Controls are very simple for that amount of work integration.

**It would be much easier to use apple shortcuts for simple controls and they would work out of the box.**

Currently I would not implement HomeKit integration due to stated issues.

Anyone else interested implementing it?

```patch
diff --git a/config.h b/config.h
index a10a31d..25fef0d 100644
--- a/config.h
+++ b/config.h
@@ -22,6 +22,13 @@
 #define CONFIG_SAVE_MAX_DELAY 10            // delay in seconds when the settings are saved after last change occured
 #define CONFIG_COMMIT_DELAY   200           // commit delay in ms

+#ifdef ENABLE_HOMEKIT_SUPPORT
+#define EEPROM_CONFIG_START 1408
+#else
+#define EEPROM_CONFIG_START 0
+#endif
+
+// size = 307 Bytes
 typedef struct {
     uint8_t brightness;
     uint8_t currentPatternIndex;
@@ -83,7 +90,7 @@ void saveConfig(bool force = false) {
     if (write_config == true) {
         SERIAL_DEBUG_LN(F("Saving Config"))
         EEPROM.begin(4095);
-        EEPROM.put(0, cfg );
+        EEPROM.put(EEPROM_CONFIG_START, cfg );
         write_config_done = true;
         write_config = false;
     }
diff --git a/esp8266-fastled-iot-webserver.ino b/esp8266-fastled-iot-webserver.ino
index ae92e70..74f762c 100644
--- a/esp8266-fastled-iot-webserver.ino
+++ b/esp8266-fastled-iot-webserver.ino
@@ -164,6 +164,8 @@ extern "C" {

     //#define ENABLE_HOMEY_SUPPORT              // Add support for Homey integration (Athom Homey library required)

+    #define ENABLE_HOMEKIT_SUPPORT
+

 //---------------------------------------------------------------------------------------------------------//

@@ -434,6 +436,24 @@ float homeySat = 1.0;           // Full
 bool  homeySolidColor = false;
 #endif
[... 146 lines cut ...]
+            HOMEKIT_CHARACTERISTIC(MANUFACTURER, LED_MANUFACTURER),
+            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0815123"),
+            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266"),
+            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, LED_VERSION),
+            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
+            NULL
+        }),
+        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
+            &cha_on,
+            &cha_name,
+            &cha_bright,
+            &cha_sat,
+            &cha_hue,
+            NULL
+        }),
+        NULL
+    }),
+    NULL
+};
+
+homekit_server_config_t accessory_config = {
+    .accessories = accessories,
+    .password = "111-11-111"
+};
```

WarDrake, 2021-02-02:
This seems not worth the work for such failure prone setup, you could however install an MQTT bridge for homekit like these :  
https://github.com/hobbyquaker/homekit2mqtt  
https://www.npmjs.com/package/homekit2mqtt  

and control it over MQTT all the payload specifications are on the README file.

bb-Ricardo, 2021-02-02:
Due to mentioned issues above I will close this issue with `wontfix`. Sorry.

closed 2021-02-02 by bb-Ricardo