NimmLor/esp8266-fastled-iot-webserver #144: Lightning pattern causes ESP exception

bb-Ricardo, 2021-01-31:
Found that the "Lightning" pattern triggers an ESP Exception and reboot.

I don't really know what causes the exeption.
```
DEBUG [50000]: Heap: 30560
DEBUG [50001]: Stats: 122 frames/s, frame delay: 7400 us
DEBUG [51006]: Stats: 101 frames/s, frame delay: 6450 us
DEBUG [52006]: Stats: 137 frames/s, frame delay: 7300 us
DEBUG [53002]: Stats: 122 frames/s, frame delay: 7400 us
DEBUG [54001]: Stats: 121 frames/s, frame delay: 7400 us
DEBUG [55000]: Stats: 121 frames/s, frame delay: 7400 us
DEBUG [56519]: Stats: 81 frames/s, frame delay: 5450 us
DEBUG [57005]: Stats: 77 frames/s, frame delay: 3300 us
DEBUG [58002]: Stats: 242 frames/s, frame delay: 9400 us
DEBUG [59009]: Stats: 98 frames/s, frame delay: 8300 us
DEBUG [60002]: Heap: 30592
DEBUG [60003]: Stats: 98 frames/s, frame delay: 7200 us
DEBUG [61003]: Stats: 124 frames/s, frame delay: 7400 us
DEBUG [62524]: Stats: 83 frames/s, frame delay: 5550 us
DEBUG [63003]: Stats: 75 frames/s, frame delay: 3300 us
DEBUG [64001]: Stats: 242 frames/s, frame delay: 9400 us
DEBUG [65008]: Stats: 98 frames/s, frame delay: 8300 us
DEBUG [66000]: Stats: 88 frames/s, frame delay: 6700 us
DEBUG [67005]: Stats: 133 frames/s, frame delay: 7350 us
DEBUG [68006]: Stats: 122 frames/s, frame delay: 7450 us
DEBUG [69004]: Stats: 120 frames/s, frame delay: 7450 us
DEBUG [70006]: Heap: 30592
DEBUG [70007]: Stats: 101 frames/s, frame delay: 6500 us
DEBUG [71003]: Stats: 135 frames/s, frame delay: 7250 us
DEBUG [72001]: Stats: 123 frames/s, frame delay: 7400 us

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40202a6e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4001f5f1 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 0190
3fffff00:  00000048 3fff107c 401006be 0c49ba5e  
3fffff10:  0000009a 00000123 00011cb5 3fff0e7d  
3fffff20:  000000d8 000000a3 3fff0024 402029d3  
3fffff30:  00000000 0000f926 3fffff20 3fffff10  
3fffff40:  00000000 3fffff30 00001388 00000004  
3fffff50:  3fff0078 3fff00e8 3fff0078 3fff0e7d  
3fffff60:  3fff0078 3fff00e8 3ffe84d4 4020ab28  
3fffff70:  00000000 41b2a8c0 80efeffe 70747400  
3fffff80:  00000000 00000000 00000001 401005d0  
3fffff90:  3fffdad0 00000000 3fff1268 3fff12a8  
3fffffa0:  3fffdad0 00000000 3fff1268 40220b58  
3fffffb0:  feefeffe feefeffe 3ffe850c 40101059  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

The Exception:
```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40202a6e: lightning() at /Users/dingsbums/Documents/Arduino/libraries/FastLED/pixeltypes.h line 169
EXCVADDR: 0x4001f5f1

Decoding stack results
0x401006be: millis() at /Users/dingsbums/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x402029d3: lightning() at /Users/dingsbums/Documents/Arduino/Meine/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 2346
0x4020ab28: loop() at /Users/dingsbums/Documents/Arduino/Meine/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1407
0x401005d0: ets_post(uint8, ETSSignal, ETSParam) at /Users/dingsbums/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40220b58: loop_wrapper() at /Users/dingsbums/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
```

As well it messes with the delay a lot.

I would propose to remove the "Lightning" pattern completely. What do you guys think?

jakesza, 2021-02-01:
I do believe this is current related. try turning down the brightness and reassess. I have a few wemos D1 mini's and thy all work with the lightning on but but reboot on an older 5v power supply.

bb-Ricardo, 2021-02-01:
This also happens without any LEDs attached. 🤔

bb-Ricardo, 2021-02-02:
Could have something to do with the amount of LEDs.

Can anyone confirm this:
* LED_DEVICE_TYPE 0
* NUM_LEDS 176
* set autoplay to off
* set pattern to Lightning

check the Serial console and see if it happens

bb-Ricardo, 2021-02-07:
Tried to reproduce and debugging it. It's not happening anymore. Strange. Who knows.

closed 2021-02-07 by bb-Ricardo