NimmLor/esp8266-fastled-iot-webserver #152: WebSocket connection fails

pattyland, 2021-02-08:
Hi,

I'm using the current in-development branch. I upload the sketch and data to my ESP. Webserver works, Wifi works, even MQTT is connecting but I can't manage anything from the web gui.

Errors from the browser console:
```
r-websocket.min.js:1 WebSocket connection to 'ws://192.168.178.55:81/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
open @ r-websocket.min.js:1
(anonymous) @ r-websocket.min.js:1
r-websocket.min.js:1 WebSocket connection to 'ws://192.168.178.55:81/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
open @ r-websocket.min.js:1
(anonymous) @ r-websocket.min.js:1
r-websocket.min.js:1 WebSocket connection to 'ws://192.168.178.55:81/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
open @ r-websocket.min.js:1
(anonymous) @ r-websocket.min.js:1
```

Sometimes there seems to be a fallback to ajax that fails too as the json is not valid:
```
Request failed: parsererror responseText: [{"name":"power","label":"Power","type":"Boolean","value":0},{"name":"brightness","label":"Brightness","type":"Number","value":255,"min":1,"max":255},{"name":"pattern","label":"Pattern","type":"Select","value":58,"options":[homeassistant/light/nanoleafs01/sethomeassistant/light/nanoleafs01/confighomeassistant/light/nanoleafs01/sethomeassistant/light/nanoleafs01/config{ "name": "Pride","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Color Waves","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Horizontal Rainbow","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Solid Rainbow","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Confetti","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Sinelon","show_palette": true,"show_speed": true,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Beat","show_palette": true,"show_speed": true,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Juggle","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Fire","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": true,"show_twinkle": false},{ "name": "Water","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": true,"show_twinkle": false},{ "name": "Strobe","show_palette": false,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Strobe","show_palette": false,"show_speed": true,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Smooth Rainbow Strobe","show_palette": false,"show_speed": true,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Roll","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Beat","show_palette": false,"show_speed": true,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Palette Fades","show_palette": true,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Chase","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Dots","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Fades","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Police Lights","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Glitter","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Snow Flakes","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Lightning","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Rainbow Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Snow Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Cloud Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Incandescent Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Retro C9 Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Red & White Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Blue & White Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Red, Green & White Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Fairy Light Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Snow 2 Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Holly Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Ice Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Party Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Forest Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Lava Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Fire Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Cloud 2 Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Ocean Twinkles","show_palette": false,"show_speed": false,"show_color_picker": false,"show_cooling_sparking": false,"show_twinkle": true},{ "name": "Solid Volume Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Static Rainbow Volume Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Flowing Rainbow Volume Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Tri-Color Volume Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Wave Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Center Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_cooling_sparking": false,"show_twinkle": false},{ "name": "Solid-Color Pair Bullet Visualizer","show_palette": true,"show_speed": true,"show_color_picker": true,"show_coo [... cut ...] y","type":"Number","value":5,"min":0,"max":8},{"name":"lines","label":"Amount of Lines for the Visualizer","type":"String","value":30},{"name":"settings","label":"Device settings","type":"Setting","value":{"deviceHostname":"nanoleafs01","otaSupport":true, "alexaSupport":false, "mqttSupport":true,"mqttEnabled":1,"mqttHostname":"192.168.178.3","mqttPort":"1883","mqttUsername":"mosquitto","mqttTopic":"homeassistant/light/nanoleafs01","mqttDevicename":"Nanoleafs"}}]
```
Any ideas?

Greets, pattyland

bb-Ricardo, 2021-02-09:
Hi Sören,

make sure you really have uploaded the correct data folder. I checked the source and could not find anything.
```shell
grep -iR "simple\|websocket" *
Changelog.md:- remove half integrated simple page #128
Changelog.md:- remove unused "webSockets" support #130
README.md:**3D-Printed IoT Bottle Lighting Pad**, a simple but smart lighting for any bottle you want.
esp8266-fastled-iot-webserver.ino:// These are old functions from previous websocket implementation
```

bb-Ricardo, 2021-02-09:
same as #151

closed 2021-02-09 by bb-Ricardo