NimmLor/esp8266-fastled-iot-webserver #155: ws2801

fabiogfabio, 2021-02-13:
Hi
i can't get the software to work if i set it as a ws2801 led strip how can i use it?

error: 

In file included from C:\Users\HWRUser\Downloads\esp8266-fastled-iot-webserver (1)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:21:0:
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
In file included from C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,
                 from C:\Users\HWRUser\Downloads\esp8266-fastled-iot-webserver (1)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:21:
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/FastLED.h:48:0,
                 from C:\Users\HWRUser\Downloads\esp8266-fastled-iot-webserver (1)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:21:
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<66u>':
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/fastspi_bitbang.h:23:50:   required from 'class AVRSoftwareSPIOutput<2u, 66u, 6u>'
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/fastspi.h:34:7:   required from 'class SPIOutput<2u, 66u, 6u>'
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/chipsets.h:158:6:   required from 'class LPD6803Controller<2u, 66u, (EOrder)10u, 6u>'
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/FastLED.h:239:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)2u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u]'
C:\Users\HWRUser\Downloads\esp8266-fastled-iot-webserver (1)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:705:68:   required from here
C:\Users\HWRUser\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Errore durante la compilazione per la scheda NodeMCU 1.0 (ESP-12E Module).

setting:

/*######################## MAIN CONFIG ########################*/
#define LED_TYPE            WS2801                     // You might also use a WS2811 or any other strip that is Fastled compatible 
#define DATA_PIN            D2                         // Be aware: the pin mapping might be different on boards like the NodeMCU
#define CLK_PIN             D0                        // Only required when using 4-pin SPI-based LEDs
#define CORRECTION          UncorrectedColor            // If colors are weird use TypicalLEDStrip
#define COLOR_ORDER         GRB                         // Change this if colors are swapped (in my case, red was swapped with green)
#define MILLI_AMPS          8000                       // IMPORTANT: set the max milli-Amps of your power supply (4A = 4000mA)
#define VOLTS               5                           // Voltage of the Power Supply

tnx!

bb-Ricardo, 2021-02-13:
Can you have a look here: https://github.com/NimmLor/esp8266-fastled-iot-webserver/issues/123 ?

seacred, 2021-08-07:
Hi 
Sadly i having the exact same problem.
I want to use a Wemos D1 mini and some WS2801.
I've already tried the solution provided in #123 but no success.
Still gettings errors such as:
`[...]\Documents\Arduino\libraries\FastLED\src/chipsets.h:158:6:   required from 'class LPD6803Controller<2, 66, RGB, 6>'
[...]\Documents\Arduino\libraries\FastLED\src/FastLED.h:239:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = WS2801; unsigned char DATA_PIN = 2; unsigned char CLOCK_PIN = 66]'
[...]\Desktop\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:692:68:   required from here
[...]\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:24: error: static assertion failed: Invalid pin specified
  210 |  static_assert(validpin(), "Invalid pin specified");
`

WarDrake, 2021-08-08:
Which version of the FastLED library are you using to compile this?

seacred, 2021-08-08:
I'm using 3.4.0

bb-Ricardo, 2021-08-11:
Hi, Did you select the correct Board in the Arduino IDE?
Which PIN config have you tried?

seacred, 2021-08-11:
Hi, actually i've tried all Wemos variants in IDE
Pins i've tried are D1+D2 like in #123 and many others but those where more like trial and error. The suggested solution in #123 sadly does not work for me with 3 different wemos d1 mini and FastLED 3.4.0.