NimmLor/esp8266-fastled-iot-webserver #159: Nanoleaf wiring direction

Roycinger, 2021-02-16:
Evening folks,
not really an issue, more a question on wiring the ws2812b strip.
Does it make a difference "which way round" the data line of the strip is wired?
2 pics for explain what I mean:
[image]
[image]

What I mean is:
Does the wiring direction in each leaf make a difference on how the finished leaf array behaves?
Thanks for clarifying!

WarDrake, 2021-02-17:
Electricly speaking, no it makes no difference at all.  
visually speaking, if you wire them all the same way SOME patterns will smoother than with random directions.  

However the effect is so minimal unless you're looking for it, you won't see it.

Roycinger, 2021-02-17:
Yes, I was speaking about the visual aspect only. Thanks for clarifying!

closed 2021-02-17 by Roycinger