NimmLor/esp8266-fastled-iot-webserver #160: Wrong Web Interface

LukasVFL99, 2021-02-19:
[image]

Why does my Web Interface look like that?
I appreciate your help.

bb-Ricardo, 2021-02-19:
Hi,

Ich weiß jetzt nicht so ganz genau was du meinst. So soll die Oberfläche aussehen.

Vermisst du was?

LukasVFL99, 2021-02-19:
Ja den color picker und die Möglichkeit die einzelnen nanoleafs anzusteuern.

Lukas Grodotzki

bb-Ricardo, 2021-02-19:
Das wurde angepasst. Bei vielen Pattern hatten die meisten Controls keine Wirkung. Es wurde so angepasst das man nur diejenigen angezeigt bekommt die auch wirklich unterstützt werden.

Hilft dir das weiter?

LukasVFL99, 2021-02-19:
Also könnte es daran liegen das ich noch keine LEDs angeschlossen habe?
Ich habe das Projekt 4 ausgewählt also die nanoleafs und wollte 8 Panels
haben mit jeweils 12 LEDs.
Bis jetzt habe ich aber leider noch keine LEDs die kommen nächste Woche
erst.

bb-Ricardo, 2021-02-21:
Dann würde das issue erst mal schließen.

Wenn weiterhin was nicht funktioniert dann kannst du gern ein neues aufmachen.

closed 2021-02-21 by bb-Ricardo