NimmLor/esp8266-fastled-iot-webserver #161: macro "SERIAL_DEBUG_ADDF" passed 2 arguments, but takes just 1

boonthegoon, 2021-02-21:
This is my first Arduino project, I was able to everything to work on the old nano leaf code and flash it but want some more features the error that I currently getting is in regards to " SERIAL_DEBUG_ADDF("failed with state %s\n", mqttClient.state())"
The error is macro"SERIAL_DEBUG_ADDF" passed 2 arguments but takes just 1. Any help will be great. thank you.            

    JsonArray effect_list = JSONencoder.createNestedArray("effect_list");
                for (uint8_t i = 0; i < patternCount; i++) {
                    effect_list.add(patterns[i].name);
                }
                size_t n = measureJson(JSONencoder);
                char mqttConfigTopic[85];
                strlcat(mqttConfigTopic, cfg.MQTTTopic, sizeof(mqttConfigTopic));
                strcat(mqttConfigTopic, "/config");
                if (mqttClient.beginPublish(mqttConfigTopic, n, true) == true) {
                    SERIAL_DEBUG_LN("Configuration Publishing Begun")
                    if (serializeJson(JSONencoder, mqttClient) == n){
                        SERIAL_DEBUG_LN("Configuration Sent")
                    }
                    if (mqttClient.endPublish() == true) {
                        SERIAL_DEBUG_LN("Configuration Publishing Finished")
                        mqttSendStatus();
                        SERIAL_DEBUG_LN("Sending Initial Status")
                    }
                } else {
                    SERIAL_DEBUG_LN("Error sending Configuration")
                }
            } else {
                SERIAL_DEBUG_ADDF("failed with state %s\n", mqttClient.state())
            }
        }
    }

    EVERY_N_SECONDS(90) {
        mqttSendStatus();
    }
#endif

boonthegoon, 2021-02-21:
this is also my first post so sorry about how it turned out

bb-Ricardo, 2021-02-21:
Hi,

thank you for reporting this issue. Indeed this is a bug in the current version. Fortunately it's easy to fix.

I pushed a new version. can you try a `git pull` and compile again?

boonthegoon, 2021-02-21:
> Hi,
> 
> thank you for reporting this issue. Indeed this is a bug in the current version. Fortunately it's easy to fix.
> 
> I pushed a new version. can you try a `git pull` and compile again?

thank you so much it worked just got another error but will try to figure it out tomorrow, This project is starting to test my patients but ill get through it, I have worked on everything but code so its a big learning curve, I read somewhere that there is a limit on how many nano leafs can be attached to one esp8266, if a signal repeater is added and power induction is done is there really a limit?

bb-Ricardo, 2021-02-21:
What error did you get?

On limit it more depends on the hardware you are using. An ESP8266 has limited RAM. You can just set the numbers of LEDs to a high value, if it still compiles it should work.
But with more LEDs the effects will get slower. On each cycle the state of the LED has to adopted/calculated.
More LEDs == more calculations per frame == less frames per second.

You can just it by setting LED_DEBUG to `1` and try out a few different LED settings.

boonthegoon, 2021-02-21:
#ifdef AddSpecificPatternDeviceA
void AlexaSpecificEventA(EspalexaDevice* d) {
    if (d == nullptr) return;
    #if LED_DEBUG != 0
    SERIAL_DEBUG_BOL
    Serial.print(" Alexa Specific Pattern update: state: "); Serial.println(d->getValue());
    #endif
    if (d->getValue() != 0)setPattern(SpecificPatternA);
    else setPattern(patternCount - 1);
}

it was on " if (d->getValue() != 0)setPattern(SpecificPatternA);"

'SpecificPatternA' was not declared in this scope

bb-Ricardo, 2021-02-21:
yes, that is a placeholder.
You need to define which pattern you want to be used when setting this Alexa event.

boonthegoon, 2021-02-21:
thank you ill try in the morning

closed 2021-02-21 by boonthegoon

blysprutan, 2021-05-14:
Hi!
I am getting the same error when i try to compile this!
"esp8266-fastled-iot-webserver:1448:79: error: macro "SERIAL_DEBUG_ADDF" passed 2 arguments, but takes just 1
                 SERIAL_DEBUG_ADDF("failed with state %s\n", mqttClient.state())
                                                                               ^
exit status 1
macro "SERIAL_DEBUG_ADDF" passed 2 arguments, but takes just 1"

I have googled and tried for hours, but i cant wrap my head around it.
i`m trying to get the lamp (device type 3) to work.
Pse help before i go nuts...

bb-Ricardo, 2021-05-14:
Hi, this should be fixed in the "in-development" branch. Can you check it out and test it please?

blysprutan, 2021-05-14:
Hi! 
I feel like total douche, but i am pretty new to all of this.. Soldering iron has been in my hand since the early 90:s but arduino is not in my ballpark.
Could you kick me in the right direction please?

bb-Ricardo, 2021-05-14:
No need to apologize, We all started at the same spot 😄.

Just open your terminal and go to your project folder. There you can switch to the dev branch:
```
git checkout in-development
```
If you you don't have/use git, this is the perfect occasion to get started with it.

Don't hesitate to ask if you get stuck.

blysprutan, 2021-05-14:
Many thanks mate! I have done it, and found it... Now i get a couple of other errors.
Should i ask here in the forum or should we chat private? or is there any other kind of chat? Discord?

bb-Ricardo, 2021-05-14:
no problem, just post them here.