NimmLor/esp8266-fastled-iot-webserver #162: 'AUDIOPATTERN' was not declared in this scope

boonthegoon, 2021-02-22:
FROM UP TOP
#ifdef ENABLE_ALEXA_SUPPORT
    #define ALEXA_DEVICE_NAME           DEFAULT_HOSTNAME
    #define AddAutoplayDevice             ((String)DEFAULT_HOSTNAME + (String)" Autoplay")
    #define AddStrobeDevice               ((String)DEFAULT_HOSTNAME + (String)" Strobe")
    #define AddSpecificPatternDeviceA     ((String)DEFAULT_HOSTNAME + (String)" Party")
    #define AddSpecificPatternDeviceB     ((String)DEFAULT_HOSTNAME + (String)" Chill")
    #define AddSpecificPatternDeviceC     ((String)DEFAULT_HOSTNAME + (String)" Rainbow")
    #define AddAudioDevice                ((String)DEFAULT_HOSTNAME + (String)" Audio")
    #define SpecificPatternA 37           // Parameter defines what pattern gets executed
    #define SpecificPatternB 12           // Parameter defines what pattern gets executed
    #define SpecificPatternC 0            // Parameter defines what pattern gets executed
    #define AudioPattern 44               // Parameter defines what pattern gets executed
#endif

WHERE ERROR WAS FOUND
#ifdef AddAudioDevice
void AlexaAudioEvent(EspalexaDevice* d) {
    if (d == nullptr) return;
    #if LED_DEBUG != 0
    SERIAL_DEBUG_BOL
    Serial.print(" Alexa Audio update: rgb: "); Serial.print(d->getR() + d->getG() + d->getB()); Serial.print(", b: "); Serial.println(d->getValue());
    #endif
    if (d->getValue() != 0)setPattern(AUDIOPATTERN);
    else setPattern(patternCount - 1);

ERROR WAS FOUND HERE "if (d->getValue() != 0)setPattern(AUDIOPATTERN);"

SHOULD "setPattern(AUDIOPATTERN)" GET CHANGED TO "setPattern(AUDIODEVICE)"

ALSO ARE THERE MORE INSTRUCTIONS THAT EXPLAIN THE ALEXA CONFIGURATIONS THAT WOULD BE GREAT

bb-Ricardo, 2021-03-03:
Hi,

This seems similar to https://github.com/NimmLor/esp8266-fastled-iot-webserver/issues/161#issuecomment-782825028

@NimmLor @WarDrake , Do you have more information on the Alexa integration.

WarDrake, 2021-03-04:
Not me, Alexa was completely done by @NimmLor I don't have any Alexa devices to test either but I'll see if I can take a look later tonight

LukasVFL99, 2021-03-05:
Just switch AUDIOPATTERN to AudioPattern this works fine

boonthegoon, 2021-03-11:
Perfect, that worked, thank you!! This is my first Arduino project, and it's starting to feel like I bit more than I can chew, so thank you for all the help and support.

closed 2021-03-11 by boonthegoon