NimmLor/esp8266-fastled-iot-webserver #166: No hardware SPI pins defined

kaliboe, 2021-02-27:
I am new with Arduino programming, I have this messages

Arduino: 1.8.8 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\arduino\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:

C:\Users\postb\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.006

 #    pragma message "FastLED version 3.002.006"

                     ^

In file included from C:\Users\postb\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\arduino\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:

C:\Users\postb\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\postb\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::DEBUG_WM(WiFiManager::wm_debuglevel_t, Generic, Genericb)':

C:\Users\postb\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:2560:9: error: 'class EspClass' has no member named 'getHeapStats'

     ESP.getHeapStats(&free, &max, &frag);

         ^

exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bb-Ricardo, 2021-02-27:
Hi and welcome to the Arduino world.

Did you make sure to use the correct library versions as described [here](https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md)?

kaliboe, 2021-02-28:
Thank you very much, I redo everything, now it works, really strange, just upgrade some libs and the I have lights working, thanks again

closed 2021-02-28 by bb-Ricardo

emersonsc, 2022-01-11:
> Hi and welcome to the Arduino world.
> 
> Did you make sure to use the correct library versions as described [here](https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md)?

I have done this, and still get the SPI pins not defined error

WarDrake, 2022-01-12:
The SPI pins not defined is not an error, it's a message and won't cause any issues with compiling or running this firmware.

whatever the error you have, is something else, try posting your log output on new issue for some more accurate help.