NimmLor/esp8266-fastled-iot-webserver #168: Blue light just blinks once and no response

xxxjajoxxx, 2021-02-28:
Hi 
I had it working once but cant connect to wifi on webserver :/ so I decided to reflash once again and all things my D1 mini do right now is blinks once blue light and no response from web serv :/ I can plug in to PC my Arduino IDE see it I can reflash it but if I trying to power it up with my ws2812b led stripe it have no response :/ all my errors on IDE 
`In file included from C:\Users\xx\xx\Dokumenty\Arduino\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:
C:\Users\xxx\xxx\Dokumenty\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from C:\Users\xxx\xxx\Dokumenty\Arduino\libraries\FastLED/FastLED.h:65:0,
                 from C:\Users\xxx\xxx\Dokumenty\Arduino\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:
C:\Users\xxx\xxx\Dokumenty\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
`

and

`Executable segment sizes:
IROM   : 389428          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27928   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1288  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 7392  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 30200 )         - zeroed variables      (global, static) in RAM/HEAP 
`

anyone had same problem?

changed hostname to different and my router still have default "LEDs"

bb-Ricardo, 2021-02-28:
Hi,

Can you switch on the LED_DEBUG option and flash it again. Leave it connected and open your Arduino serial monitor. There you should see what's going on.

xxxjajoxxx, 2021-02-28:
⸮⸮8NL⸮⸮q⸮H⸮⸮⸮8NL⸮⸮⸮HH⸮8⸮J⸮⸮⸮NL⸮NŜ⸮15:01:40.626 -> ⸮⸮?NL⸮H⸮⸮⸮H⸮⸮JŜ⸮⸮⸮H⸮⸮HH⸮(⸮6J⸮⸮H|⸮⸮8NL⸮⸮8NL⸮H⸮0!H⸮⸮⸮⸮H⸮9⸮H⸮⸮⸮⸮HŜ⸮H⸮⸮⸮)⸮HŜ⸮H⸮H⸮⸮8NL⸮⸮0⸮⸮⸮H⸮⸮H<⸮ȏ⸮

dont know random marks :/

bb-Ricardo, 2021-02-28:
This looks like it's working as supposed to be.

Can you hit the reset button on the D1 and post whats printed to serial Monitor from boot.

Did you connect to the Wifi and select your wifi and entered the credentials?

xxxjajoxxx, 2021-02-28:
`DEBUG [187]: Loading config

DEBUG [188]: System Information:
DEBUG [188]: Version: 4.5 (2020-02-14)
DEBUG [189]: Heap: 39232
DEBUG [189]: SDK: 2.2.2-dev(38a443e)
DEBUG [190]: Boot Vers: 31
DEBUG [193]: CPU Speed: 80 MHz
DEBUG [195]: Chip ID: 90950
DEBUG [198]: Flash ID: 1458415
DEBUG [200]: Flash Size: 4194304KB
DEBUG [203]: Vcc: 65535
DEBUG [206]: MAC address: 48:3F:DA:01:63:46

*WM: [1] getCoreVersion():          2_7_0
*WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             37032
*WM: [1] AutoConnect 
*WM: [3] setupHostname:  LEDs
*WM: [2] Setting WiFi hostname 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: xxx
*WM: [3] Using Password: xxx
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  LEDs - 6346
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             LEDs - 6346
*WM: [1] password:         
*WM: [1] ssid_len:         11
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2182 ms
*WM: [2] Config Portal Running, non blocking/processing 
INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP 'LEDs - 6346' to configure your wireless connection
DEBUG [12236]: SPIFFS contents:
DEBUG [12245]: FS File: /css/bootstrap.min.css, size: 121093
DEBUG [12245]: FS File: /css/dark-mode.css, size: 1350
DEBUG [12253]: FS File: /css/minicolors.min.css, size: 96597
DEBUG [12254]: FS File: /css/styles.css, size: 1392
DEBUG [12259]: FS File: /favicon.ico, size: 4286
DEBUG [12264]: FS File: /fonts/glyphicons.woff2, size: 18028
DEBUG [12268]: FS File: /images/atom196.png, size: 3889
DEBUG [12273]: FS File: /images/github.ico, size: 6518
DEBUG [12278]: FS File: /images/surr.png, size: 1897
DEBUG [12283]: FS File: /index.htm, size: 9706
DEBUG [12287]: FS File: /js/app.js, size: 16150
DEBUG [12292]: FS File: /js/bootstrap.min.js, size: 37045
DEBUG [12296]: FS File: /js/dark-mode-switch.min.js, size: 492
DEBUG [12305]: FS File: /js/jquery-3.1.1.min.js, size: 86709
DEBUG [12307]: FS File: /js/minicolors.min.js, size: 15463
DEBUG [12312]: FS File: /settings.htm, size: 10129

DEBUG [12318]: FS Size: 957314KB, used: 438999KB, 45.86%

DEBUG [12321]: Arduino Core Version: 2_7_0
DEBUG [12325]: Enabled Features:
DEBUG [12328]: Feature: mDNS support enabled

DEBUG [12331]: Configured device type: LED strip (0)
DEBUG [12336]: NUM_LEDS: 150
DEBUG [12339]: BAND_GROUPING: 25
DEBUG [12341]: PACKET_LENGTH: 6
INFO: HTTP web server started
DEBUG [13011]: Stats: 45 frames/s, frame delay: 4250 us
DEBUG [14006]: Stats: 99 frames/s, frame delay: 3200 us
DEBUG [15005]: Stats: 111 frames/s, frame delay: 2750 us
DEBUG [16004]: Stats: 117 frames/s, frame delay: 2600 us

`

bb-Ricardo, 2021-02-28:
Hi,

```
INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP 'LEDs - 6346' to configure your wireless connection
```

This is the important information. Did you do that?

And you might need to try different LED_PIN settings to find the correct PIN.

Cause even in the access point mode the lights should work out of the box.

xxxjajoxxx, 2021-03-02:
sorry my bad :/ I realized it few minutes after i write this post, i forget this newer version dont have config in arduino to auto connect with wifi

closed 2021-03-02 by xxxjajoxxx