NimmLor/esp8266-fastled-iot-webserver #170: ESP32 - Guru Meditation Error

Pajaentr, 2021-03-03:
Hello, I have installed the latest version of your code on ESP32 (NodeMCU-32S) and sometimes when I change the configuration I get this error:
`22:16:48.841 -> DEBUG [5000]: Stats: 125 frames/s, frame delay: 5850 us
22:16:49.816 -> DEBUG [6001]: Stats: 125 frames/s, frame delay: 6100 us
22:16:50.189 -> DEBUG [6364]: Setting: speed: 96
22:16:50.189 -> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
22:16:50.189 -> Core 1 register dump:
22:16:50.189 -> PC      : 0xa9e0400e  PS      : 0x00060a30  A0      : 0x800d90c1  A1      : 0x3ffb1dd0  
22:16:50.235 -> A2      : 0x3ffc23ec  A3      : 0x3ffd272c  A4      : 0xa9e0400e  A5      : 0x3ffb1e20  
22:16:50.235 -> A6      : 0x3ffc3e24  A7      : 0xff000000  A8      : 0x800eb3c6  A9      : 0x3ffb1db0  
22:16:50.235 -> A10     : 0x3ffc23ec  A11     : 0x3ffd272c  A12     : 0x00000053  A13     : 0x000000b4  
22:16:50.235 -> A14     : 0x00000002  A15     : 0x3ffc3e24  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
22:16:50.235 -> EXCVADDR: 0xa9e0400c  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  
22:16:50.235 -> 
22:16:50.235 -> ELF file SHA256: 0000000000000000
22:16:50.281 -> 
22:16:50.281 -> Backtrace: 0x69e0400e:0x3ffb1dd0 0x400d90be:0x3ffb1e00 0x400d95f2:0x3ffb1e20 0x400d995f:0x3ffb1e50 0x400e0c4e:0x3ffb1e90 0x400ebc51:0x3ffb1eb0 0x400ebcbe:0x3ffb1ee0 0x400ebe3d:0x3ffb1f30 0x400d8cd1:0x3ffb1f80 0x400ef3a9:0x3ffb1fb0 0x40089d8a:0x3ffb1fd0
22:16:50.281 -> 
`
and after inserting this information to "Exception Decoder" I get this:
`PC: 0xa9e0400e
EXCVADDR: 0xa9e0400c

Decoding stack results
0x400d90be: sendString(String) at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1638
0x400d95f2: sendInt(unsigned char) at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1627
0x400d995f: std::_Function_handler   >::_M_invoke(const std::_Any_data &) at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1191
0x400e0c4e: std::function ::operator()() const at d:\soubory\google drive\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400ebc51: FunctionRequestHandler::handle(WebServer&, HTTPMethod, String) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\detail/RequestHandlersImpl.h line 45
0x400ebcbe: WebServer::_handleRequest() at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 635
0x400ebe3d: WebServer::handleClient() at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 314
0x400d8cd1: loop() at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1346
0x400ef3a9: loopTask(void*) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 37
0x40089d8a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

My settings:
`#define LED_TYPE            WS2812B              
#define DATA_PIN            27                          
//#define CLK_PIN             D5                        
#define CORRECTION          UncorrectedColor
#define COLOR_ORDER         GRB
#define MILLI_AMPS          10000
#define VOLTS               5
#define LED_DEBUG 1 
#define DEFAULT_HOSTNAME "LEDs"
#define LED_DEVICE_TYPE 0`

bb-Ricardo, 2021-03-03:
Hi,

this looks like a duplicate of #163 .

Can you try to run in in Debug mode as suggested in that issue.

I don't have an ESP32 so it makes it hard to debug.

Pajaentr, 2021-03-03:
**I hope this will help you to find the problematic piece of code. I had to edit "boards.txt" file to enable debug/verbose settings.
The problem now appeared when trying to connect via Chrome on a laptop.**

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:07:14.322 -> configsip: 0, SPIWP:0xee
00:07:14.322 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:07:14.322 -> mode:DIO, clock div:1
00:07:14.322 -> load:0x3fff0018,len:4
00:07:14.322 -> load:0x3fff001c,len:1044
00:07:14.322 -> load:0x40078000,len:10124
00:07:14.322 -> load:0x40080400,len:5856
00:07:14.322 -> entry 0x400806a8
00:07:14.792 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
00:07:14.792 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
00:07:14.885 -> 
00:07:14.885 -> 
00:07:14.885 -> DEBUG [265]: Loading config
00:07:14.931 -> 
00:07:14.931 -> DEBUG [277]: System Information:
00:07:14.931 -> DEBUG [277]: Version: 4.5 (2020-02-14)
00:07:14.931 -> DEBUG [277]: Heap: 202152
00:07:14.931 -> DEBUG [277]: SDK: master
00:07:14.931 -> DEBUG [277]: CPU Speed: 240 MHz
00:07:14.931 -> DEBUG [279]: Flash Size: 4194304KB
00:07:14.931 -> DEBUG [282]: MAC address: 3C:61:05:15:94:5C
00:07:14.931 -> 
00:07:15.023 -> *WM: [1] AutoConnect 
00:07:15.023 -> *WM: [2] ESP32 event handler enabled 
00:07:15.212 -> *WM: [2] Connecting as wifi client... 
00:07:15.212 -> *WM: [2] setSTAConfig static ip not set, skipping 
00:07:15.260 -> *WM: [1] Connecting to SAVED AP: UPC7720214
00:07:15.725 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
00:07:15.818 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
00:07:15.818 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
00:07:15.818 -> [D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.0.52, MASK: 255.255.255.0, GW: 192.168.0.1
00:07:15.818 -> *WM: [2] Connection result: WL_CONNECTED
00:07:15.865 -> *WM: [1] AutoConnect: SUCCESS 
00:07:15.865 -> *WM: [1] STA IP Address: 192.168.0.52
00:07:15.865 -> INFO: Wi-Fi connected
00:07:15.865 -> DEBUG [1199]: SPIFFS contents:
00:07:15.959 -> DEBUG [1313]: FS File: /css/bootstrap.min.css, size: 121093
00:07:15.959 -> DEBUG [1314]: FS File: /css/dark-mode.css, size: 1350
00:07:15.959 -> DEBUG [1328]: FS File: /css/minicolors.min.css, size: 96597
00:07:15.959 -> DEBUG [1329]: FS File: /css/styles.css, size: 1392
00:07:15.959 -> DEBUG [1330]: FS File: /favicon.ico, size: 4286
00:07:16.006 -> DEBUG [1334]: FS File: /fonts/glyphicons.woff2, size: 18028
00:07:16.006 -> DEBUG [1337]: FS File: /images/atom196.png, size: 3889
00:07:16.006 -> DEBUG [1342]: FS File: /images/github.ico, size: 6518
00:07:16.006 -> DEBUG [1346]: FS File: /images/surr.png, size: 1897
00:07:16.006 -> DEBUG [1351]: FS File: /index.htm, size: 9706
00:07:16.006 -> DEBUG [1356]: FS File: /js/app.js, size: 16150
00:07:16.006 -> DEBUG [1364]: FS File: /js/bootstrap.min.js, size: 37045
00:07:16.006 -> DEBUG [1364]: FS File: /js/dark-mode-switch.min.js, size: 492
00:07:16.006 -> DEBUG [1380]: FS File: /js/jquery-3.1.1.min.js, size: 86709
00:07:16.006 -> DEBUG [1382]: FS File: /js/minicolors.min.js, size: 15463
00:07:16.052 -> DEBUG [1384]: FS File: /settings.htm, size: 10129
00:07:16.052 -> 
00:07:16.052 -> DEBUG [1401]: FS Size: 1378241KB, used: 438999KB, 31.85%
00:07:16.052 -> 
00:07:16.052 -> DEBUG [1401]: Enabled Features:
00:07:16.052 -> DEBUG [1401]: Feature: mDNS support enabled
00:07:16.052 -> DEBUG [1401]: Feature: OTA support enabled
00:07:16.052 -> DEBUG [1405]: Feature: UDP visualization support enabled
00:07:16.052 -> 
00:07:16.052 -> DEBUG [1410]: Configured device type: LED strip (0)
00:07:16.052 -> DEBUG [1414]: NUM_LEDS: 48
00:07:16.052 -> DEBUG [1417]: BAND_GROUPING: 4
00:07:16.052 -> DEBUG [1419]: PACKET_LENGTH: 12
00:07:16.146 -> [V][RequestHandlersImpl.h:72] StaticRequestHandler(): StaticRequestHandler: path=/ uri=/ isFile=0, cache_header=max-age=86400
00:07:16.191 -> 
[... 41 lines cut ...]
00:07:28.779 -> Core 1 register dump:
00:07:28.779 -> PC      : 0x00000073  PS      : 0x00060630  A0      : 0x800ec996  A1      : 0x3ffb1d80  
00:07:28.826 -> A2      : 0x3ffc24dc  A3      : 0x3ffd276c  A4      : 0x3ffd2530  A5      : 0x3ffb1df0  
00:07:28.826 -> A6      : 0x3ffc3f54  A7      : 0xff000000  A8      : 0x800ec936  A9      : 0x3ffb1d60  
00:07:28.826 -> A10     : 0x3ffc24dc  A11     : 0x3ffd276c  A12     : 0x00000081  A13     : 0x000000b4  
00:07:28.826 -> A14     : 0x00000000  A15     : 0x3ffc3f54  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
00:07:28.826 -> EXCVADDR: 0x00000070  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  
00:07:28.826 -> 
00:07:28.826 -> ELF file SHA256: 0000000000000000
00:07:28.826 -> 
00:07:28.826 -> Backtrace: 0x00000073:0x3ffb1d80 0x400ec993:0x3ffb1db0 0x400eca9d:0x3ffb1dd0 0x400ececa:0x3ffb1e20 0x400ed2da:0x3ffb1ee0 0x400ed491:0x3ffb1f30 0x400d9225:0x3ffb1f80 0x400f1981:0x3ffb1fb0 0x40089dc2:0x3ffb1fd0

**And text from "Exception Decoder":**
PC: 0x00000073
EXCVADDR: 0x00000070

Decoding stack results
0x400ec993: WebServer::send(int, String const&, String const&) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 465
0x400eca9d: WebServer::_streamFileCore(unsigned int, String const&, String const&) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 524
[... cut ...]
0x400ed491: WebServer::handleClient() at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 314
0x400d9225: loop() at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1346
0x400f1981: loopTask(void*) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 37
0x40089dc2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Pajaentr, 2021-03-04:
If you need any more data to debug, I'll be happy to provide it.

bb-Ricardo, 2021-03-04:
Did you try different versions of the ESP32 library?

bb-Ricardo, 2021-03-04:
Hey, can you try this commit which I just pushed?

Pajaentr, 2021-03-05:
**Still the same error**
20:55:45.378 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:55:45.378 -> configsip: 0, SPIWP:0xee
20:55:45.378 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:55:45.378 -> mode:DIO, clock div:1
20:55:45.378 -> load:0x3fff0018,len:4
20:55:45.378 -> load:0x3fff001c,len:1044
20:55:45.378 -> load:0x40078000,len:10124
20:55:45.378 -> load:0x40080400,len:5856
20:55:45.378 -> entry 0x400806a8
20:55:45.851 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
20:55:45.851 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
20:55:45.945 -> 
20:55:45.945 -> 
20:55:45.945 -> DEBUG [264]: Loading config
20:55:45.992 -> 
20:55:45.992 -> DEBUG [276]: System Information:
20:55:45.992 -> DEBUG [277]: Version: 4.5 (2020-02-14)
20:55:45.992 -> DEBUG [277]: Heap: 205160
20:55:45.992 -> DEBUG [277]: SDK: master
20:55:45.992 -> DEBUG [277]: CPU Speed: 240 MHz
20:55:45.992 -> DEBUG [279]: Flash Size: 4194304KB
20:55:45.992 -> DEBUG [282]: MAC address: 3C:61:05:15:94:5C
20:55:45.992 -> 
20:55:46.087 -> *WM: [1] AutoConnect 
20:55:46.087 -> *WM: [2] ESP32 event handler enabled 
20:55:46.277 -> *WM: [2] Connecting as wifi client... 
20:55:46.277 -> *WM: [2] setSTAConfig static ip not set, skipping 
20:55:46.277 -> *WM: [1] Connecting to SAVED AP: Pavel
20:55:46.796 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
20:55:46.890 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
20:55:46.890 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
20:55:46.890 -> [D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.106, MASK: 255.255.255.0, GW: 192.168.1.1
20:55:46.984 -> *WM: [2] Connection result: WL_CONNECTED
20:55:46.984 -> *WM: [1] AutoConnect: SUCCESS 
20:55:46.984 -> *WM: [1] STA IP Address: 192.168.1.106
20:55:46.984 -> INFO: Wi-Fi connected
20:55:46.984 -> DEBUG [1293]: SPIFFS contents:
20:55:47.125 -> DEBUG [1407]: FS File: /css/bootstrap.min.css, size: 121093
20:55:47.125 -> DEBUG [1409]: FS File: /css/dark-mode.css, size: 1350
20:55:47.125 -> DEBUG [1423]: FS File: /css/minicolors.min.css, size: 96597
20:55:47.125 -> DEBUG [1424]: FS File: /css/styles.css, size: 1392
20:55:47.125 -> DEBUG [1425]: FS File: /favicon.ico, size: 4286
20:55:47.125 -> DEBUG [1429]: FS File: /fonts/glyphicons.woff2, size: 18028
20:55:47.125 -> DEBUG [1432]: FS File: /images/atom196.png, size: 3889
20:55:47.125 -> DEBUG [1437]: FS File: /images/github.ico, size: 6518
20:55:47.173 -> DEBUG [1441]: FS File: /images/surr.png, size: 1897
20:55:47.173 -> DEBUG [1446]: FS File: /index.htm, size: 9706
20:55:47.173 -> DEBUG [1451]: FS File: /js/app.js, size: 16150
20:55:47.173 -> DEBUG [1459]: FS File: /js/bootstrap.min.js, size: 37045
20:55:47.173 -> DEBUG [1460]: FS File: /js/dark-mode-switch.min.js, size: 492
20:55:47.173 -> DEBUG [1475]: FS File: /js/jquery-3.1.1.min.js, size: 86709
20:55:47.173 -> DEBUG [1477]: FS File: /js/minicolors.min.js, size: 15463
20:55:47.173 -> DEBUG [1479]: FS File: /settings.htm, size: 10129
20:55:47.173 -> 
20:55:47.173 -> DEBUG [1496]: FS Size: 1378241KB, used: 438999KB, 31.85%
20:55:47.219 -> 
20:55:47.219 -> DEBUG [1496]: Enabled Features:
20:55:47.219 -> DEBUG [1497]: Feature: mDNS support enabled
20:55:47.219 -> 
20:55:47.219 -> DEBUG [1497]: Configured device type: LED strip (0)
20:55:47.219 -> DEBUG [1501]: NUM_LEDS: 48
20:55:47.219 -> DEBUG [1503]: BAND_GROUPING: 4
20:55:47.219 -> DEBUG [1506]: PACKET_LENGTH: 12
20:55:47.267 -> [V][RequestHandlersImpl.h:72] StaticRequestHandler(): StaticRequestHandler: path=/index.htm uri=/ isFile=1, cache_header=max-age=86400
20:55:47.267 -> 
20:55:47.408 -> [V][RequestHandlersImpl.h:72] StaticRequestHandler(): StaticRequestHandler: path=/css/bootstrap.min.css uri=/css/bootstrap.min.css isFile=1, cache_header=max-age=86400
20:55:47.408 -> 
20:55:47.408 -> [V][RequestHandlersImpl.h:72] StaticRequestHandler(): StaticRequestHandler: path=/css/dark-mode.css uri=/css/dark-mode.css isFile=1, cache_header=max-age=86400
20:55:47.408 -> 
[... 63 lines cut ...]
20:55:51.644 -> A2      : 0x3ffc231c  A3      : 0x3ffd2260  A4      : 0x3ffb1e60  A5      : 0x3ffb1df0  
20:55:51.644 -> A6      : 0x3ffc36f4  A7      : 0xff000000  A8      : 0x800eae0a  A9      : 0x3ffb1d60  
20:55:51.644 -> A10     : 0x3ffc231c  A11     : 0x3ffd2260  A12     : 0x00000073  A13     : 0x000000b4  
20:55:51.690 -> A14     : 0x00000000  A15     : 0x3ffc36f4  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
20:55:51.690 -> EXCVADDR: 0x0ea37040  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  
20:55:51.690 -> 
20:55:51.690 -> ELF file SHA256: 0000000000000000
20:55:51.690 -> 
20:55:51.690 -> Backtrace: 0x0ea37040:0x3ffb1d80 0x400eae67:0x3ffb1db0 0x400eaf71:0x3ffb1dd0 0x400eb39e:0x3ffb1e20 0x400eb7ae:0x3ffb1ee0 0x400eb965:0x3ffb1f30 0x400d911e:0x3ffb1f80 0x400eef21:0x3ffb1fb0 0x40089dc2:0x3ffb1fd0

**Log from decoder**
PC: 0x0ea37040
EXCVADDR: 0x0ea37040

Decoding stack results
0x400eae67: WebServer::send(int, String const&, String const&) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 465
0x400eaf71: WebServer::_streamFileCore(unsigned int, String const&, String const&) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 524
[... cut ...]
0x400eb965: WebServer::handleClient() at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\libraries\WebServer\src\WebServer.cpp line 314
0x400d911e: loop() at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver1/esp8266-fastled-iot-webserver1.ino line 1357
0x400eef21: loopTask(void*) at D:\Soubory\Google Drive\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 37
0x40089dc2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

**Now I use ESP32 library version 1.0.5. Tomorrow I can try version 1.0.4.**

bb-Ricardo, 2021-03-05:
Darn. I don't really know what we're doing wrong here.

Pajaentr, 2021-03-05:
I tried one more thing - clear the memory with this command:
_esptool --chip esp32 --port com11 erase_flash_
and flash the code again.

After this process, everything worked for a while, but then ESP32 started restarting again.

Here is link to log:
[https://drive.google.com/file/d/1BQLc0sr5KAkFdb-IMyfxXfjucoEUm6a4/view?usp=sharing](url)

Pajaentr, 2021-03-08:
I am sorry, it took much more time than I expected, but today I tested ESP32 library version 1.0.4.
I clean memory with command abowe, flash the program and after a while the program crashed again

**Here is the log**
[esp32 lib 1.0.4 log.txt](https://github.com/NimmLor/esp8266-fastled-iot-webserver/files/6103764/esp32.lib.1.0.4.log.txt)

**Decode of first crash**
PC: 0x7561223d
EXCVADDR: 0x00000000

Decoding stack results
0x400ea46f: WebServer::send(int, String const&, String const&) at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src\WebServer.cpp line 453
0x400ea579: WebServer::_streamFileCore(unsigned int, String const&, String const&) at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src\WebServer.cpp line 509
0x400ea89a: StaticRequestHandler::handle(WebServer&, HTTPMethod, String) at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h line 138
0x400eaada: WebServer::_handleRequest() at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src\WebServer.cpp line 620
0x400eac43: WebServer::handleClient() at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src\WebServer.cpp line 309
0x400d8952: loop() at D:\Soubory\Google Drive\3D Printer\Tisk\Nanoleaf\Firmware\esp8266-fastled-iot-webserver1/esp8266-fastled-iot-webserver1.ino line 1355
0x400eda19: loopTask(void*) at C:\Users\pajae\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40089299: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

bb-Ricardo, 2021-03-08:
Hi, thank you for all this effort.

It has something to do with reading a file and sending the file content to the client.
But to make any actual progress on this I need an ESP32 myself.

I just ordered a bunch but it will take a while for them to get here.

Will let you know as soon as I have more code to test. Sorry.

bb-Ricardo, 2021-03-25:
Hey,

Just wanted to give you a quick update.

Got my delivery two days ago and tried to figure out what the issue is.

I'm stumped. I get errors on simple functions. The error is not persistent, get "Guru mediation error" from different parts of the code.

Also the performance on frames per second is !! lower !! then a regular ESP8266. Also automatically adjusting the FPS does not work as expected.

Not sure what to make out of all of this except remove the ESP32 support and mark it as "NOT WORKING".

Also read about CPU pinning some tasks as the ESP32 has two CPUs, one for your network traffic handling and the other one for your code.

Anyone else thoughts on this?

WarDrake, 2021-03-25:
I ordered some ESP32's as well and will test when they get here, but it sounds a bit like the NodeMCU boards I use, had to create my own branch to stop random errors and messages from appearing because of the interrupts implementation.

that might be somethign you might want to test, see my fork for the NodeMCU branch I run