NimmLor/esp8266-fastled-iot-webserver #171: OTA does not work

Roycinger, 2021-03-04:
Hi,
although I configured the OTA module parameters, the upload does not work.
I'm always getting prompted for the OTA password (although it's set).
Additionally, I get a "No response from device" message.
[image]

Does anyone deal with a similar problem?
Thanks!

bb-Ricardo, 2021-03-04:
Does it work without password?

Roycinger, 2021-03-04:
> Does it work without password?

Unfortuantely not

bb-Ricardo, 2021-03-04:
Did you start the OTA mode via GUI?

Roycinger, 2021-03-04:
> Did you start the OTA mode via Web interface?

Yes. The LEDs freeze, UI unavailable (as per design), the ESP gets displayed in the Arduino IDE under ports.
As soon as the sketch is compiled and the actual upload begins, I'm getting the described behaviour

bb-Ricardo, 2021-03-04:
I'm sorry. I actually don't know how to debug this. What do you get in your serial console when you try to perform a OTA Update?

Roycinger, 2021-03-17:
> I'm sorry. I actually don't know how to debug this. What do you get in your serial console when you try to perform a OTA Update?

Sorry, I didnt see the last notification.
Serial Monitor is not available when doing an OTA update

obenschlaefer, 2023-11-05:
[image]
 
 you can find a password optin in line 151 (maybe +-5 lines in your code) 
 
 Just try to find it by using the search option in arduino ID (strg+f)