NimmLor/esp8266-fastled-iot-webserver #172: Error in 2 patterns, using nano leaf mode, plz remove Sound reactive mode from autoplay

DrMucki, 2021-03-07:
Hi first of all : A great JOB! 
I have minor suggestions...

When using the nanolaef mode (I used 10 Leafs with each 29 LEDs) I found the following errors in pattern
In "Pride" and "color Waves" 9 out of 10. leafs work perfect. The last one only works with only a part of the leds. (maybe 8 or 10) the rest is not synchron, showing solid or different colors. That has nothing to do with voltage dropping ...

The I have the suggestion to leave the sound reactive mode out of the Autoplay

WarDrake, 2021-03-07:
The problem you're seeing is because of the 29 LEDs per leaf, several of the patters are setup as 1/3rd of the leaf LEDs to play with the sides or corners as an animation, 29 isn't divisible by 3 and will give you issues in those patterns.

DrMucki, 2021-03-07:
should have known this before... then I would have printed them larger.. but 30 didn't fit..
what about my suggestion?

bb-Ricardo, 2021-03-07:
Skipping sound reactive pattern during autoplay is a great idea. Will add this soon.