NimmLor/esp8266-fastled-iot-webserver #174: FS.h: No such file or directory

KingsRequin, 2021-03-11:
hello, I'm still starting on the esp I'm sorry
I have this error:
Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "Arduino Mini, ATmega328P"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:mini:cpu=atmega328 -ide-version=10813 -build-path C:\Users\Megaport\AppData\Local\Temp\arduino_build_463779 -warnings=none -build-cache C:\Users\Megaport\AppData\Local\Temp\arduino_cache_224540 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose E:\Documents\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:mini:cpu=atmega328 -ide-version=10813 -build-path C:\Users\Megaport\AppData\Local\Temp\arduino_build_463779 -warnings=none -build-cache C:\Users\Megaport\AppData\Local\Temp\arduino_cache_224540 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose E:\Documents\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

Using board 'mini' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MINI -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\Megaport\\AppData\\Local\\Temp\\arduino_build_463779\\sketch\\esp8266-fastled-iot-webserver.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for FastLED.h: [FastLED@3.4.0]

ResolveLibrary(FastLED.h)

  -> candidates: [FastLED@3.4.0]

"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MINI -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IE:\\Documents\\Arduino\\libraries\\FastLED\\src" "C:\\Users\\Megaport\\AppData\\Local\\Temp\\arduino_build_463779\\sketch\\esp8266-fastled-iot-webserver.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for FS.h: []

ResolveLibrary(FS.h)

  -> candidates: []esp8266-fastled-iot-webserver:22:10: fatal error: FS.h: No such file or directory

 #include <FS.h>

          ^~~~~~

compilation terminated.

Utilisation de la bibliothèque FastLED version 3.4.0 dans le dossier: E:\Documents\Arduino\libraries\FastLED 

exit status 1

FS.h: No such file or directory

However, it seems to me to have followed the tutorial well ...
thanks !

WarDrake, 2021-03-11:
This means you're missing the fastLED library, you probably skipped that one, or it might have failed to install properly.
just make sure you install that library and it should compile.

KingsRequin, 2021-03-21:
Yes thanks ! That was it ! but now I have this problem that I can not solve ...
```
Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Megaport\AppData\Local\Temp\arduino_build_667291 -warnings=none -build-cache C:\Users\Megaport\AppData\Local\Temp\arduino_cache_908536 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=E:\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=E:\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=E:\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=E:\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.python3.path=E:\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=E:\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=E:\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=E:\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose E:\Documents\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Megaport\AppData\Local\Temp\arduino_build_667291 -warnings=none -build-cache C:\Users\Megaport\AppData\Local\Temp\arduino_cache_908536 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=E:\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=E:\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=E:\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=E:\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.python3.path=E:\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=E:\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=E:\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=E:\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose E:\Documents\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

Using board 'd1_mini' from platform in folder: E:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4

Using core 'esp8266' from platform in folder: E:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4

Detecting libraries used...

"E:\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IE:\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IE:\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IE:\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Megaport\\AppData\\Local\\Temp\\arduino_build_667291/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=8 [... cut ...]
Utilisation de la bibliothèque FastLED version 3.4.0 dans le dossier: E:\Documents\Arduino\libraries\FastLED 

Utilisation de la bibliothèque EEPROM version 1.0 dans le dossier: E:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\EEPROM 

Utilisation de la bibliothèque ESP8266mDNS version 1.2 dans le dossier: E:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS 

exit status 1

Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini```

closed 2021-04-11 by KingsRequin