NimmLor/esp8266-fastled-iot-webserver #177: None in mqtt payload

Toriniasty, 2021-03-27:
Hi,
When using mqtt integration it sends payload like this to mqtt:
```
{"~":"homeassistant/light/nanoleaf_1","name":"NanoLeaf_1","dev":{"ids":"84:CC:A8:AA:0E:34","mf":"Surrbradl08","mdl":"0.4","name":"NanoLeaf_1"},"stat_t":"~","cmd_t":"~/set","brightness":true,"rgb":true,"effect":true,"uniq_id":"84:CC:A8:AA:0E:34","schema":"json","effect_list":["Pride","Color Waves","Horizontal Rainbow","Solid Rainbow","Confetti","Sinelon","Beat","Juggle","Fire","Water","Strobe","Rainbow Strobe","Smooth Rainbow Strobe","Rainbow Roll","Rainbow Beat","Palette Fades","Rainbow Chase","Rainbow Fades","Police Lights","Glitter","Snow Flakes","Lightning","Rainbow Twinkles","Snow Twinkles","Cloud Twinkles","Incandescent Twinkles","Retro C9 Twinkles","Red & White Twinkles","Blue & White Twinkles","Red, Green & White Twinkles","Fairy Light Twinkles","Snow 2 Twinkles","Holly Twinkles","Ice Twinkles","Party Twinkles","Forest Twinkles","Lava Twinkles","Fire Twinkles","Cloud 2 Twinkles","Ocean Twinkles","Solid Volume Visualizer","Static Rainbow Volume Visualizer","Flowing Rainbow Volume Visualizer","Tri-Color Volume Visualizer","Wave Visualizer","Center Visualizer","Solid-Color Pair Bullet Visualizer","Solid-Color Complementary Bullet Visualizer","Blue/Purple Bullet Visualizer","Beat-Bullet Visualization","Bass Ring Visualizer","Kick Ring Visualizer","Rainbow Band Visualizer",null]}
```

Which is still valid JSON, however in Home Assistant integration via MQTT we get this error:
```
2021-03-18 10:36:57 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'name': 'NanoLeaf_1', 'device': {'identifiers': '84:CC:A8:AA:0E:34', 'manufacturer': 'Surrbradl08', 'model': '0.4', 'name': 'NanoLeaf_1'}, 'state_topic': 'homeassistant/light/nanoleaf_1', 'command_topic': 'homeassistant/light/nanoleaf_1/set', 'brightness': True, 'rgb': True, 'effect': True, 'unique_id': '84:CC:A8:AA:0E:34', 'schema': 'json', 'effect_list': ['Pride', 'Color Waves', 'Horizontal Rainbow', 'Solid Rainbow', 'Confetti', 'Sinelon', 'Beat', 'Juggle', 'Fire', 'Water', 'Strobe', 'Rainbow Strobe', 'Smooth Rainbow Strobe', 'Rainbow Roll', 'Rainbow Beat', 'Palette Fades', 'Rainbow Chase', 'Rainbow Fades', 'Police Lights', 'Glitter', 'Snow Flakes', 'Lightning', 'Rainbow Twinkles', 'Snow Twinkles', 'Cloud Twinkles', 'Incandescent Twinkles', 'Retro C9 Twinkles', 'Red & White Twinkles', 'Blue & White Twinkles', 'Red, Green & White Twinkles', 'Fairy Light Twinkles', 'Snow 2 Twinkles', 'Holly Twinkles', 'Ice Twinkles', 'Party Twinkles', 'Forest Twinkles', 'Lava Twinkles', 'Fire Twinkles', 'Cloud 2 Twinkles', 'Ocean Twinkles', 'Solid Volume Visualizer', 'Static Rainbow Volume Visualizer', 'Flowing Rainbow Volume Visualizer', 'Tri-Color Volume Visualizer', 'Wave Visualizer', 'Center Visualizer', 'Solid-Color Pair Bullet Visualizer', 'Solid-Color Complementary Bullet Visualizer', 'Blue/Purple Bullet Visualizer', 'Beat-Bullet Visualization', 'Bass Ring Visualizer', 'Kick Ring Visualizer', 'Rainbow Band Visualizer', None], 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 152, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 25, in validate_mqtt_light
    return schemas[value[CONF_SCHEMA]](value)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: string value is None @ data['effect_list'][53]
```
While this is is still valid JSON, I was wondering if there's any way to fix it for HA?

WarDrake, 2021-03-28:
The message length got cut off, seems like you're using an old version of the code, this was solved some time ago, try updating to the newest release from the releases page

WarDrake, 2021-04-08:
Did this solve your problem?

WarDrake, 2021-07-09:
Closing this due to lack of a response, open a new issue if there's somethig else.

closed 2021-07-09 by WarDrake