NimmLor/esp8266-fastled-iot-webserver #182: Flickering when using any of the bar type visualizations - anything bigger than an 8x8 matrix

aceofjokers, 2021-04-08:
Hey All,

Running into issues with any of the bar type vizualizations - rainbow, outside, solid colour....where there are random flickers across the matrix whenever the matrix is anything bigger than 8x8.

Video below shows this - thats an 8x16 matrix with config being 8x24

have tried on NodeMCU and D1 mini

https://user-images.githubusercontent.com/6286590/113967922-6a7b6780-9875-11eb-9639-f41cd9ea6952.mp4

Any help/ideas appreciated.

WarDrake, 2021-04-08:
It looks to me like there's too much draw, or the fps might be too low... 
have you tried injecting power on the second matrix? 

I'm running 126 leds to your 128 and my 5 amp power source sometimes has a bit of trouble with some of the patterns.
try the debug mode, you can see the fps there and let us know what it is, and try injecting power on the second matrix see if that solves it

aceofjokers, 2021-04-12:
Hey @WarDrake thanks for responding.

I've tried injecting additional power into the boards, as well as usb direct to the ESP with the same issue persisting.

For what its worth, all other patterns work great, and can use max brightness white across 3x 8x8 matrix's with no color change across the lot.

screenshot of debugging FPS + what is set in the code:
[image]
Is it supposed to be this far off? 
I've tried 240fps and 60fps with no changes.

The 'flicker' still occurs even with no music playing - when the AudioVisualizer software is running.

https://user-images.githubusercontent.com/6286590/114336099-935f6d80-9b91-11eb-9019-43ba1a010f0c.mp4

Thanks for your help!

WarDrake, 2021-04-12:
It's not supposed to be that far off
[image]
however, it might be because of the amount of leds, since each lead is an instruction sent per cycle.

That flicker on the video is not about FPS tho.

I see the UDP messages there, meaning you're using the audio visualizer to control the leds, I think that's where the Issue is coming from, it's in Beta and only @NimmLor really would know what the issue there might be...

aceofjokers, 2021-04-28:
Haha cheers! One of the guys at work turned around and said - You should make something like this! https://www.amazon.com.au/LIFX-International-Adjustable-Multicolour-Assistant/dp/B07543MM1Y

I found this repo and it was a challenge accepted. Still a while to go, but getting there.

Awesome, good to know (fps).

Not quite sure what made it work, but here it is again! (back to a nodemcu over the mini) with a few extra boards, powered the same way it was before.

https://user-images.githubusercontent.com/6286590/116405575-29261880-a873-11eb-9587-61ee3f411265.mp4

Ill mark this down to....."re-do it all again and it might work!".

Cheers for your help @bb-Ricardo !!

closed 2021-04-28 by aceofjokers

bb-Ricardo, 2021-04-28:
Wow, that looks great.

Glad it all worked out. Maybe ground our the signal line was not 100% proper conected and caused this issue.

Enjoy