NimmLor/esp8266-fastled-iot-webserver #183: Urls to control Lamp directly

L3-N0X, 2021-04-11:
I really like the work, I have 5 lamps working with this code. But if I want to change the color of them, I must go on every IP adress and change the pattern. So I thought of coding an Android App, which can control every lamp with one Button.

But to get this to work, i need a link or something similar which pushes the On and Off button automatically. Is there a way to get such a link? 

For example:  http://<Ip adress>/power=0

If not, is there an other way to control the light without the user pushing the Button on the website?

L3-N0X, 2021-04-11:
After trying a few things, I got it to work with a HTTP Request Widget using this Setup:

[image]
[image]

closed 2021-04-11 by L3-N0X

bb-Ricardo, 2021-04-11:
Great. There should be a detailed list of url params in the Readme.