NimmLor/esp8266-fastled-iot-webserver #187: Cant open the server via a browser

eMapleLuke, 2021-04-24:
I set my data for my home wifi, but i cant go into the web interface.

bb-Ricardo, 2021-04-24:
Hi, 
Just try to switch on LED_DEBUG and check what the serial monitor prints out.

eMapleLuke, 2021-04-24:
And how can i trun it on?

eMapleLuke, 2021-04-24:
The Esp also dosnt epar in my Fritz.box

bb-Ricardo, 2021-04-26:
great, so it is working good now?

eMapleLuke, 2021-04-26:
Yes, but my Alexa don't find the ESP. Do I have to change something in the code?

bb-Ricardo, 2021-04-26:
you have to activate the Alexa Feature as described in the docs.

eMapleLuke, 2021-04-26:
But when I turn on the features a error message show up.

aceofjokers, 2021-04-28:
Whats the error?

eMapleLuke, 2021-04-29:
Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:

C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:

C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void AlexaSpecificEventA(EspalexaDevice*)':

esp8266-fastled-iot-webserver:3951:39: error: 'SpecificPatternA' was not declared in this scope

     if (d->getValue() != 0)setPattern(SpecificPatternA);

                                       ^

C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void AlexaSpecificEventB(EspalexaDevice*)':

esp8266-fastled-iot-webserver:3959:39: error: 'SpecificPatternB' was not declared in this scope

     if (d->getValue() != 0)setPattern(SpecificPatternB);

                                       ^

C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void AlexaSpecificEventC(EspalexaDevice*)':

esp8266-fastled-iot-webserver:3967:39: error: 'SpecificPatternC' was not declared in this scope

     if (d->getValue() != 0)setPattern(SpecificPatternC);

                                       ^

Mehrere Bibliotheken wurden für "ArduinoOTA.h" gefunden

 Benutzt: C:\Users\lukeb\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ArduinoOTA

 Nicht benutzt: C:\Users\lukeb\Documents\Arduino\libraries\ArduinoOTA

exit status 1

'SpecificPatternA' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

bb-Ricardo, 2021-04-29:
Hi,

Der Wert `SpecificPatternA` ist nur ein Platzhalter. Du musst hier das Pattern eintragen was durch das Alexa Kommando getriggert werden soll.

eMapleLuke, 2021-04-29:
Ok und wie genau mache ich das?

bb-Ricardo, 2021-04-30:
Schau mal hier:
https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/386efc6d168b074213e9cb01b7ebdb9e09de6f0f/esp8266-fastled-iot-webserver.ino#L551-L669

Da sind die Namen der Pattern in Anführungszeichen.

Das könnte dann so aussehen:
```c
 if (d->getValue() != 0)setPattern("Horizontal Rainbow");
```

Versuchs mal.

eMapleLuke, 2021-04-30:
Ok, ich habe dann jetzt versucht das über OTA zu updaten, aber nachdem ich das Passwort eingegeben habe zeigt mir das kurz OK an und danach sofort, Host antwortet nicht. Dann hab ich das nochmal mit Kabel versucht und dann kam nur diese Fehlermeldung: Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:

C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\lukeb\Desktop\Neuer Ordner (2)\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:

C:\Users\lukeb\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

Executable segment sizes:

IROM   : 423712          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 28700   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1360  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 9060  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 29480 )         - zeroed variables      (global, static) in RAM/HEAP 

Der Sketch verwendet 462832 Bytes (44%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.

Globale Variablen verwenden 39900 Bytes (48%) des dynamischen Speichers, 42020 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.

esptool.py v2.8

Serial port COM3

Traceback (most recent call last):

  File "C:\Users\lukeb\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/lukeb/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2889, in main

    esp = chip_class(each_port, initial_baud, args.trace)

  File "C:/Users/lukeb/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 237, in __init__

    self._port = serial.serial_for_url(port)

  File "C:/Users/lukeb/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\__init__.py", line 88, in serial_for_url

    instance.open()

  File "C:/Users/lukeb/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\serialwin32.py", line 62, in open

    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))

serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Zugriff verweigert', None, 5)

serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Zugriff verweigert', None, 5)

bb-Ricardo, 2021-04-30:
Hast du den Port im Arduono wieder zurück gestellt? Da hilft leider auch manchmal einfach den Rechner neu zu starten.

Bei OTA einfach mal die Passwort option auskommentieren. Hat zumindest bei mir immer gut geklappt.

eMapleLuke, 2021-04-30:
Also per OTA geht das einfach nicht, wenn ich das mit dem Passwort auskommentiere fragt das trotzdem noch nach einem. Also hab das mit dem Kabel dann einfach wieder auf COM 3 gestellt. 
Aber warum auch immer leuchten bei mir die LEDs nur in einem zufälligen Muster und kann diese auch nicht steuern. Nur wenn ich den ESP neu starte ändert sich was, kann die aber dann immernoch nicht steuern.

bb-Ricardo, 2021-05-01:
Dann müsstest du mal die Ausgabe des Serial Monitors hier posten.

eMapleLuke, 2021-05-01:
Ok konnte jetzt die neue Software drauf laden, kann die LEDs aber immer noch nicht steuern.

bb-Ricardo, 2021-05-01:
Ich helfe gern aber dazu dazu benötige ich die Serielle Ausgabe. Kann dir sonst wirklich nicht helfen.

eMapleLuke, 2021-05-01:
Ich sehe da aber nur komische Zeichen.

bb-Ricardo, 2021-05-01:
Tur mir leid aber ich werde dir hier nicht weiter helfen können und werde auch nicht mehr kommentieren und antworten.

Sorry, I refuse to offer any further help. This is leading nowhere.