NimmLor/esp8266-fastled-iot-webserver #189: Alexa doesn't find device

Roycinger, 2021-05-06:
Hi folks,

after my Wemos went up in smoke for whatever reason, I flashed a new Wemos with the arduino sketch.
I enabled Alexa support, made the corresponding settings in the sketch, Debug also shows me that Alexa Support is enabled:
[image]

In general, the Nanoleaf clone is working fine (apart from the OTA feature which never worked for me) but my Echo Dot is unable to find the device. Also restarting the Dot didnt work out.

Is there anything else I can debug or investigate in?

Thanks!

emersonsc, 2022-01-12:
Having the same issue. My Echo Dot and the phone app say they didnt find a new device

zarigata, 2022-02-22:
me as well.... any updated?

xbavnu, 2022-03-05:
I am having the same problem. Did anyone find a solution?

Also I can not use MQTT. I see that it is connecting to MQTT every couple seconds, but I can not control it via MQTT.

WarDrake, 2022-03-25:
Please check with new version

Maschine2501, 2022-05-29:
Same issue here, just installed newest versions right from github...
Here's the Log of the device discovery:
`Espalexa Begin...
MAXDEVICES 10
Done
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: ssdp:all
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: upnp:rootdevice
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: ssdp:all
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: upnp:rootdevice
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: upnp:rootdevice
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: ssdp:all
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: upnp:rootdevice
MAN: "ssdp:discover"
MX: 3

Responding search req...
Got UDP!
Got UDP!
# Responding to description.xml ... #

Send setup.xml
<?xml version="1.0" ?><root xmlns="urn:schemas-upnp-org:device-1-0"><specVersion><major>1</major><minor>0</minor></specVersion><URLBase>http://192.168.178.119:80/</URLBase><device><deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType><friendlyName>Espalexa (192.168.178.119:80)</friendlyName><manufacturer>Royal Philips Electronics</manufacturer><manufacturerURL>http://www.philips.com</manufacturerURL><modelDescription>Philips hue Personal Wireless Lighting</modelDescription><modelName>Philips hue bridge 2012</modelName><modelNumber>929000226503</modelNumber><modelURL>http://www.meethue.com</modelURL><serialNumber>c45bbe5579f4</serialNumber><UDN>uuid:2f402f80-da50-11e1-9b23-c45bbe5579f4</UDN><presentationURL>index.html</presentationURL></device></root>
# Responding to description.xml ... #

Send setup.xml
[... 468 lines cut ...]
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
`

PS: Awesome Work! Thank you!