NimmLor/esp8266-fastled-iot-webserver #191: Some problem for MQTT and ALEXA support...

hank102223, 2021-05-12:
(Sorry,my English isn't well because my country don't always use English.)

if i ENABLE_MQTT_SUPPORT (line=165)

[image]

Arduino says:
macro "SERIAL_DEBUG_ADDF" passed 2 arguments, but takes just 1
(line=1448)

and if i ENABLE_ALEXA_SUPPORT (line=159)

[image]

Arduino says:
'AUDIOPATTERN' was not declared in this scope
(line=4443)

how to fix it?
Thanks.

bb-Ricardo, 2021-05-14:
Hi,

Your english seems great, nothing to worry about.

The first bug has been fixed in the "in-development" branch. Can you check out this branch and compile it again?

The second one is just a placeholder. You need to select an available pattern.

Have a look here:
https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/386efc6d168b074213e9cb01b7ebdb9e09de6f0f/esp8266-fastled-iot-webserver.ino#L551-L669

These are the pattern names in quotes:

This could look like this for example:
```c
 if (d->getValue() != 0)setPattern("Horizontal Rainbow");
```

Can you try that please?

hank102223, 2021-05-14:
[image]
everything is ok in development branch
thank you

and it is my old sketch file
there is some problem on it,just for you to reference.

[esp8266-fastled-iot-webserver.zip](https://github.com/NimmLor/esp8266-fastled-iot-webserver/files/6478543/esp8266-fastled-iot-webserver.zip)

WarDrake, 2022-03-25:
Check on the newest version please

Boerny41, 2022-05-05:
Since the "in-development" branch is 15 month old by now I just wanted to check back if we should still use it when getting the MQTT error.

[image]

Boerny41, 2022-05-05:
It was completly my fault I'm sorry. I went to the in-development branch and hit the download button on the right not realizing that it would download the main reposetory this way.

Now with the correct files everything seems to work fine.