NimmLor/esp8266-fastled-iot-webserver #192: Some Errors

HolyVaping, 2021-05-17:
Hi, im quite new to this and i wanted to do the Nano Leaf Replica.

But im getting massive errors and got no clue what todo, im close to run berserk.

Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:

C:\Users\mbikk\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Users\mbikk\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:

C:\Users\mbikk\OneDrive\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void setup()':

esp8266-fastled-iot-webserver:703:5: error: 'default_cfg' was not declared in this scope

     default_cfg.brightness = brightness; 

     ^

esp8266-fastled-iot-webserver:753:32: error: 'cfg' was not declared in this scope

     String nameString = String(cfg.hostname) + String(" - ") + macID;

                                ^

C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In lambda function:

esp8266-fastled-iot-webserver:1023:51: error: 'cfg' is not captured

         json += "{\"deviceHostname\":\"" + String(cfg.hostname) + "\"";

                                                   ^

C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In lambda function:

esp8266-fastled-iot-webserver:1063:39: error: 'setWiFiConf' was not declared in this scope

             setWiFiConf(ssid, password);

                                       ^

esp8266-fastled-iot-webserver:1069:50: error: 'cfg' is not captured

         if (new_hostname.length() != 0 && String(cfg.hostname) != new_hostname) {

                                                  ^

esp8266-fastled-iot-webserver:1070:37: error: 'setHostname' was not declared in this scope

             setHostname(new_hostname);

                                     ^

esp8266-fastled-iot-webserver:1119:28: error: 'saveConfig' was not declared in this scope

             saveConfig(true);

[... 158 lines cut ...]
     setConfigChanged();

                      ^

C:\Users\mbikk\Downloads\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void setSpeed(uint8_t)':

esp8266-fastled-iot-webserver:1852:5: error: 'cfg' was not declared in this scope

     cfg.speed = speed;

     ^

esp8266-fastled-iot-webserver:1853:22: error: 'setConfigChanged' was not declared in this scope

     setConfigChanged();

                      ^

exit status 1

'default_cfg' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

bb-Ricardo, 2021-05-17:
Hi, did you do a fresh clone? It seems there are some files missing. For example `config.h`.

HolyVaping, 2021-05-17:
I used got one out of the package. What do i need to do with a fresh Wemos D1 Mini?

HolyVaping, 2021-05-17:
Können auch auf Deutsch schreiben geht wahrscheinlich einfacher. Ich habe alles mehrmals so gemacht wie in der Anleitung beschrieben, kriege aber diese Fehlermeldungen.

HolyVaping, 2021-05-17:
Habe es denke ich mal jetzt irgendwie hin bekommen. Arduino hat eine Datei einfach verschoben.

HolyVaping, 2021-05-17:
Jetzt habe ich das Problem nachdem ich mein WLAN etc eingestellt habe, er unter der angezeigten IP in der Monitor Konsole nichts findet, obwohl er in der Fritzbox etc. alles anzeigt. Eine Idee wodran es liegen kann=

bb-Ricardo, 2021-05-17:
Welche IP zeigt er denn an?

HolyVaping, 2021-05-17:
INFO: Wi-Fi connected
INFO: HTTP web server started
INFO: WiFi Connected! Open http://192.168.179.5 in your browser
INFO: mDNS responder started. Try to open http://LEDs.local in your browser

HolyVaping, 2021-05-17:
Ok es lag wohl daran das ich den Gastzugang benutzt habe und da wohl was blockiert wurde. Werde Morgen mal das Nano Leaf zusammenbauen und löten, danach berichten.

HolyVaping, 2021-05-17:
Da ich jetzt das dritte Mini board "benutzt" habe. Kann man die anderen 2 reseten oder kann ich diese "entsorgen"? Schönen Abend noch

bb-Ricardo, 2021-05-18:
Prima das es doch noch geklappt hat. Die anderen Boards sollten auch funktionieren. Da gibt es wohl in der Arduino IDE eine Option wie man auch die daten im EEPROM auf "0" setzt.