NimmLor/esp8266-fastled-iot-webserver #194: Help a Noob - First LED Flickering

MrPresident2, 2021-06-07:
Hey Guys! 
This noob needs your help.
So my first LED is flickering a bit and when I turn the LEDs off through the website interface everything shut off except for the first LED that continues to blink/flicker. I changed the entire led strip and it's still happening.
Help!
Thanks :)

WarDrake, 2021-06-07:
Can you get a log of the serial output while with debug enabled so we can help you diagnose the issue?

MrPresident2, 2021-06-07:
> Can you get a log of the serial output while with debug enabled so we can help you diagnose the issue?

Hey Sir! 
Thank your help.
I can 100% do what you wrote up there if you woke me through the steps. As I said I'm a noob in this world and trying to learn :)

WarDrake, 2021-06-08:
You need to set the constant called : LED_DEBUG to 1 and then upload the firmware.
then run start the arduino while it is still connected to the computer and use a terminal to look at the output and just save it and paste it here

WarDrake, 2021-06-08:
There doesn't seem to be any issue with the software here, might be something with your physical device or connections, did you wire the data line of the led directly to the arduino? try adding a resistor in there if you did, or if not, maybe check if a different board does the same thing, software wise everyhing looks right

MrPresident2, 2021-06-08:
I don't have a different board for now but I do have a few coming in in a few weeks, what kind of resistor should I use before the LED? My data line right now is connected directly to the board

WarDrake, 2021-06-08:
It's probably noise from the data line then, anything between 100 to 1k resistor will do the trick of keeping the line clean

WarDrake, 2021-06-08:
In reality it doesn't matter at all, but I tend to work with 1/4 as those are the easiest ones to get a hold of for me.

MrPresident2, 2021-06-09:
I just added the resistor between the data and the led but the first led is still flickering. so it didnt work

WarDrake, 2021-06-09:
If you still have noise on the data line with a resistor there then I can only speculate there's some other hardware fault as the software is reporting no issues at all, double check your connections, your power supply and if not then try another board when you get one.

is your power connected directly to the led or is it running through the board first and then the leds? can you show your wiring diagram?

MrPresident2, 2021-06-10:
The power connects straight to the LED, I guess ill have to wait for the new boards to come

a-ayad, 2021-06-23:
I have also the exact same issue, I have connected a resistor and replaced the board and the problem is still there. If I try a normal code to light up the LEDs, the flickering is gone.

jchilders99, 2021-06-24:
This is an issue with the esp8266 v3.0.0 library.  Roll back to the 2.7.4 library and the issue will go away.  I haven't opened an issue with them if someone wants to take the lead ;)

MrPresident2, 2021-06-24:
> This is an issue with the esp8266 v3.0.0 library. Roll back to the 2.7.4 library and the issue will go away. I haven't opened an issue with them if someone wants to take the lead ;)

OMG that worked! Thanks!

jchilders99, 2021-06-24:
> > This is an issue with the esp8266 v3.0.0 library. Roll back to the 2.7.4 library and the issue will go away. I haven't opened an issue with them if someone wants to take the lead ;)
> 
> OMG that worked! Thanks!

No problem!  It was super annoying and I swear I was getting micro flickers in the LEDs that has also gone away with 2.7.4.  I swear the animations are smoother with the older library.

WarDrake, 2021-06-24:
Will close this Issue then, thanks for chiming in @jchilders99

closed 2021-06-24 by WarDrake