NimmLor/esp8266-fastled-iot-webserver #196: API Post "not Found"

sn1ce, 2021-07-08:
Hi there,

first of all, very cool project, i use this in like 5 lights currently.

I just wanted to know the correct api post request for solidColor..
Just red the code, but can't manage to find the correct syntax for it.

For the pattern it was easy

/pattern?value=56 (solid Color)

But when i try to use the solidColor in my POST (using curl in bash)
/solidColor?255,255,255

it says: not found.
Even in my webbrowser..

Can you guys maybe give a hint or a well documented version of all api requests?

Much appreciated!
Best,
sn1ce

WarDrake, 2021-07-08:
The correct address to call is : 
/solidColor?r=255&g=255&b=255

as for pattern, use : 
/patternName?value=Solid%20Color

the 56 is dependent on the number of patters available and/or visualizations so it might not be the same across different installations or configurations and might change on version updates. (mine is 57)

while using patternName will always yield the expected one

sn1ce, 2021-07-10:
Hi again,

i tested your call and it's alos saying "not found".

I am using

bb-Ricardo, 2021-07-11:
Hi,

Good point. Some examples should be added to the documentation.