NimmLor/esp8266-fastled-iot-webserver #197: cannot access webpage

ve3sjk, 2021-07-10:
I am using the in development branch since the release gave me the error on debug message about passing too many parameters.

I have the software flashed into 2 different d1 mini units and only get timeouts when trying to go to the webpages of the unit. This is the debug output of the first unit, the second is the same with the exception of the IP address being different. I followed the instructions to the T and did the spiff flash as well but neither unit is accessible via webpage in both case the matrix turns to all light blue and does nothing else. 

Can you point me to where i did something wrong in the setup to cause this.

`12:29:14.985 -> DEBUG [191]: Loading config
12:29:14.985 -> 
12:29:14.985 -> DEBUG [192]: System Information:
12:29:14.985 -> DEBUG [192]: Version: 4.5 (2020-02-14)
12:29:14.985 -> DEBUG [192]: Heap: 36592
12:29:14.985 -> DEBUG [193]: SDK: 2.2.2-dev(38a443e)
12:29:14.985 -> DEBUG [194]: Boot Vers: 6
12:29:14.985 -> DEBUG [196]: CPU Speed: 80 MHz
12:29:14.985 -> DEBUG [199]: Chip ID: 13638900
12:29:14.985 -> DEBUG [202]: Flash ID: 1458392
12:29:14.985 -> DEBUG [204]: Flash Size: 4194304KB
12:29:14.985 -> DEBUG [207]: Vcc: 65535
12:29:14.985 -> DEBUG [209]: MAC address: 3C:61:05:D0:1C:F4
12:29:14.985 -> 
12:29:15.038 -> *wm:[1] AutoConnect 
12:29:15.038 -> *wm:[2] Setting WiFi hostname 
12:29:15.038 -> *wm:[2] Connecting as wifi client... 
12:29:15.038 -> *wm:[2] setSTAConfig static ip not set, skipping 
12:29:15.038 -> *wm:[1] Connecting to SAVED AP: graveyard1
12:29:15.566 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
12:29:17.441 -> *wm:[2] Connection result: WL_CONNECTED
12:29:17.441 -> *wm:[1] AutoConnect: SUCCESS 
12:29:17.441 -> *wm:[1] STA IP Address: 192.168.1.245
12:29:17.441 -> INFO: Wi-Fi connected
12:29:17.441 -> DEBUG [2578]: SPIFFS contents:
12:29:17.441 -> DEBUG [2586]: FS File: /css/bootstrap.min.css, size: 121093
12:29:17.441 -> DEBUG [2587]: FS File: /css/dark-mode.css, size: 1350
12:29:17.441 -> DEBUG [2594]: FS File: /css/minicolors.min.css, size: 96597
12:29:17.441 -> DEBUG [2596]: FS File: /css/styles.css, size: 1392
12:29:17.441 -> DEBUG [2600]: FS File: /favicon.ico, size: 4286
12:29:17.441 -> DEBUG [2605]: FS File: /fonts/glyphicons.woff2, size: 18028
12:29:17.441 -> DEBUG [2610]: FS File: /images/atom196.png, size: 3889
12:29:17.441 -> DEBUG [2614]: FS File: /images/github.ico, size: 6518
12:29:17.441 -> DEBUG [2619]: FS File: /images/surr.png, size: 1897
12:29:17.441 -> DEBUG [2624]: FS File: /index.htm, size: 9706
12:29:17.441 -> DEBUG [2628]: FS File: /js/app.js, size: 16150
12:29:17.441 -> DEBUG [2633]: FS File: /js/bootstrap.min.js, size: 37045
12:29:17.441 -> DEBUG [2636]: FS File: /js/dark-mode-switch.min.js, size: 492
12:29:17.441 -> DEBUG [2645]: FS File: /js/jquery-3.1.1.min.js, size: 86709
12:29:17.441 -> DEBUG [2648]: FS File: /js/minicolors.min.js, size: 15463
12:29:17.441 -> DEBUG [2652]: FS File: /settings.htm, size: 10129
12:29:17.488 -> 
12:29:17.488 -> DEBUG [2679]: FS Size: 1953282KB, used: 438999KB, 22.47%
12:29:17.488 -> 
12:29:17.488 -> DEBUG [2680]: Arduino Core Version: 
12:29:17.488 -> DEBUG [2681]: Enabled Features:
12:29:17.488 -> DEBUG [2681]: Feature: mDNS support enabled
12:29:17.488 -> DEBUG [2683]: Feature: MQTT support enabled (mqtt version: 4)
12:29:17.488 -> 
12:29:17.488 -> DEBUG [2689]: Configured device type: LED MATRIX (1)
12:29:17.488 -> DEBUG [2693]: NUM_LEDS: 360
12:29:17.488 -> DEBUG [2696]: BAND_GROUPING: 1
12:29:17.488 -> DEBUG [2698]: PACKET_LENGTH: 30
12:29:17.542 -> INFO: HTTP web server started
12:29:17.788 -> DEBUG [3005]: Stats: 11 frames/s, frame delay: 2550 us
12:29:17.788 -> INFO: WiFi Connected! Open http://192.168.1.245 in your browser
12:29:17.788 -> INFO: mDNS responder started. Try to open http://LEDs.local in your browser`

ve3sjk, 2021-07-11:
I gave up, time to look for a better solution. I tried this software on 4 different D1 Mini boards, all of which work perfectly with matrixGFX demo and single string demos. This software compiles and flashes fine, connects to internet but most of the boards will not connect to the webpages and none of them would do anything but flash the 1st led in the string. I did get 1 board to give me access to the webpage but it had no effect and was not controlling anything. As far as I am concerned this software does not work on the D1 mini board i tried setting up for led matrix which did not work i even tried the same board with a generic setting for a 300 led string and it did not work, all tests where via the board and the leds running off a 20 amp 5 volt supply. I flashed matixGFX demo back into all the boards and everyone of them run the matrix as expected. There is  something wrong with this firmware on my boards. Don't know what it is wasted two days trying to get this to run.  Going to need a different way to drive the nanoleafs i have printed.

craigjmidwinter, 2021-08-19:
I'm assuming you found another solution? I'm seeing the same thing. Logs say the server started but can't access the page

craigjmidwinter, 2021-08-19:
Ok, I don't know if this is related, but I also had MQTT enabled but not configured and saw this issue. I noticed there was a bunch of code for both MQTT and the webserver in close proximity to each other, so I disabled the MQTT feature and reflashed and the web server came up @NimmLor

craigjmidwinter, 2021-08-19:
Maybe I spoke too soon. It came up, but after having it unplugged and then plugging it back in, its inaccessible. Logs still show the webserver starting

WarDrake, 2021-08-19:
Have you tried erasing all flash memory and then reuploading the data?

craigjmidwinter, 2021-08-19:
Yeah, I tried that. I just tried updating this block in the wifimanager setup to set the autoreconnect: 
```
  if (wifiManager.autoConnect(nameChar)) {
    Serial.println("INFO: Wi-Fi connected");
    Serial.println("INFO: Setting autoreconnect");
    wifiManager.setWiFiAutoReconnect(true);
  } else {
    Serial.printf("INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP '%s' to configure your wireless connection\n", nameChar);
    wifiMangerPortalRunning = true;
  }
```
Not sure if that is actually doing anything, but it seems stable right now. I'll keep an eye on it for a few days and if it seems to make things better I can make a PR

craigjmidwinter, 2021-08-22:
Looks like my change did not improve things. It seems like it gets a connection on init and immediately drops it. Enough to get an IP address, but it never becomes pingable

craigjmidwinter, 2021-08-24:
I've tried this on two separate boards now, and both have the exact same behaviour

WarDrake, 2021-08-24:
can you post the changes you made to the configuration of the script?

craigjmidwinter, 2021-08-24:
Yep for sure. This is my device configuration section:
```
//---------------------------------------------------------------------------------------------------------//
// Device Configuration:
//---------------------------------------------------------------------------------------------------------//
#if LED_DEVICE_TYPE == 0                // Generic LED-Strip
#define NUM_LEDS 300
//#define NUM_LEDS 33
//#define NUM_LEDS 183
#define BAND_GROUPING    1            // Groups part of the band to save performance and network traffic
#elif LED_DEVICE_TYPE == 1              // LED MATRIX
#define LENGTH 32
#define HEIGHT 8
//#define AddLogoVisualizers          // (only 32x8) Adds Visualization patterns with logo (currently only HBz)
#elif LED_DEVICE_TYPE == 2              // 7-Segment Clock
#define NTP_REFRESH_INTERVAL_SECONDS 600            // 10 minutes
const char* ntpServerName = "at.pool.ntp.org";      // Austrian ntp-timeserver
int t_offset = 1;                                   // offset added to the time from the ntp server
bool updateColorsEverySecond = false;               // if set to false it will update colors every minute (time patterns only)
const int NTP_PACKET_SIZE = 48;
bool switchedTimePattern = true;
#define NUM_LEDS 30
#define Digit1 0
#define Digit2 7
#define Digit3 16
#define Digit4 23
// Values for the Big Clock: 58, 0, 14, 30, 44

#elif LED_DEVICE_TYPE == 3              // Desk Lamp
#define LINE_COUNT    8             // Amount of led strip pieces
#define LEDS_PER_LINE 10            // Amount of led pixel per single led strip piece

#elif LED_DEVICE_TYPE == 4              // Nanoleafs
#define LEAFCOUNT 8                // Amount of triangles
#define PIXELS_PER_LEAF 36          // Amount of LEDs inside 1x Tringle

#elif LED_DEVICE_TYPE == 5              // Animated Logos
// Choose your logo below, remove the comment in front of your design
// Important: see "LOGO CONFIG" below

#define TWENTYONEPILOTS
//#define THINGIVERSE     // FIXME: THIS IS BROKEN

#endif
```
I was strongly thinking it was a network issue since it looked like it would get an IP and not respond to pings when this happened, but I enabled and configured MQTT, and that works as expected even when the server doesn't respond to pings or http requests

bb-Ricardo, 2021-09-02:
This is indeed very strange. And you used the recommended library versions as mentioned here?: https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md#dependencies

craigjmidwinter, 2021-09-03:
Yep, same library versions

bb-Ricardo, 2021-09-03:
And if you switch to option 1 (LED Strip) to test it. Dame behavior?

craigjmidwinter, 2021-09-07:
Yeah, same behaviour on option 1. Here's the output
```
15:34:00.847 -> DEBUG [180]: Loading config
15:34:00.847 -> 
15:34:00.847 -> DEBUG [181]: System Information:
15:34:00.847 -> DEBUG [181]: Version: 4.5 (2020-02-14)
15:34:00.847 -> DEBUG [182]: Heap: 37904
15:34:00.847 -> DEBUG [182]: SDK: 2.2.2-dev(38a443e)
15:34:00.880 -> DEBUG [183]: Boot Vers: 31
15:34:00.880 -> DEBUG [186]: CPU Speed: 80 MHz
15:34:00.880 -> DEBUG [188]: Chip ID: 5567178
15:34:00.880 -> DEBUG [191]: Flash ID: 1458392
15:34:00.880 -> DEBUG [194]: Flash Size: 4194304KB
15:34:00.880 -> DEBUG [197]: Vcc: 65535
15:34:00.880 -> DEBUG [199]: MAC address: C4:5B:BE:54:F2:CA
15:34:00.880 -> 
15:34:00.914 -> *WM: [1] getCoreVersion():          2_7_4
15:34:00.914 -> *WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
15:34:00.947 -> *WM: [1] system_get_boot_version(): 31
15:34:00.947 -> *WM: [1] getFreeHeap():             35688
15:34:00.947 -> *WM: [1] AutoConnect 
15:34:00.947 -> *WM: [1] No Credentials are Saved, skipping connect 
15:34:00.947 -> *WM: [2] Starting Config Portal 
15:34:00.947 -> *WM: [3] WIFI station disconnect 
15:34:00.947 -> *WM: [3] WiFi station enable 
15:34:00.947 -> *WM: [2] Disabling STA 
15:34:00.947 -> *WM: [2] Enabling AP 
15:34:00.947 -> *WM: [1] StartAP with SSID:  test-leafs - F2CA
15:34:01.448 -> *WM: [2] AP has anonymous access! 
15:34:01.448 -> *WM: [1] SoftAP Configuration 
15:34:01.448 -> *WM: [1] -------------------- 
15:34:01.448 -> *WM: [1] ssid:             test-leafs - F2CA
15:34:01.481 -> *WM: [1] password:         
15:34:01.481 -> *WM: [1] ssid_len:         17
15:34:01.481 -> *WM: [1] channel:          1
15:34:01.481 -> *WM: [1] authmode:        
15:34:01.481 -> *WM: [1] ssid_hidden:     
15:34:01.481 -> *WM: [1] max_connection:   4
15:34:01.481 -> *WM: [1] country:          CN
15:34:01.481 -> *WM: [1] beacon_interval:  100(ms)
15:34:01.481 -> *WM: [1] -------------------- 
15:34:01.981 -> *WM: [1] AP IP address: 192.168.4.1
15:34:01.981 -> *WM: [3] setupConfigPortal 
15:34:01.981 -> *WM: [1] Starting Web Portal 
15:34:01.981 -> *WM: [3] dns server started with ip:  192.168.4.1
15:34:01.981 -> *WM: [2] HTTP server started 
15:34:04.181 -> *WM: [2] WiFi Scan completed in 2184 ms
15:34:04.181 -> *WM: [2] Config Portal Running, non blocking/processing 
15:34:04.181 -> INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP 'test-leafs - F2CA' to configure your wireless connection
15:34:04.181 -> DEBUG [3505]: SPIFFS contents:
15:34:04.181 -> DEBUG [3508]: FS File: /index.htm, size: 9706
15:34:04.181 -> DEBUG [3512]: FS File: /favicon.ico, size: 4286
15:34:04.181 -> DEBUG [3516]: FS File: /css/styles.css, size: 1392
15:34:04.214 -> DEBUG [3520]: FS File: /css/dark-mode.css, size: 1350
15:34:04.214 -> DEBUG [3530]: FS File: /css/bootstrap.min.css, size: 121093
15:34:04.214 -> DEBUG [3535]: FS File: /css/minicolors.min.css, size: 96597
15:34:04.214 -> DEBUG [3536]: FS File: /settings.htm, size: 10129
15:34:04.214 -> DEBUG [3540]: FS File: /images/atom196.png, size: 3889
15:34:04.214 -> DEBUG [3545]: FS File: /images/github.ico, size: 6518
15:34:04.214 -> DEBUG [3549]: FS File: /images/surr.png, size: 1897
15:34:04.247 -> DEBUG [3554]: FS File: /fonts/glyphicons.woff2, size: 18028
15:34:04.247 -> DEBUG [3560]: FS File: /js/minicolors.min.js, size: 15463
15:34:04.247 -> DEBUG [3568]: FS File: /js/jquery-3.1.1.min.js, size: 86709
15:34:04.247 -> DEBUG [3569]: FS File: /js/dark-mode-switch.min.js, size: 492
15:34:04.247 -> DEBUG [3575]: FS File: /js/app.js, size: 16150
15:34:04.247 -> DEBUG [3580]: FS File: /js/bootstrap.min.js, size: 37045
15:34:04.281 -> 
15:34:04.281 -> DEBUG [3607]: FS Size: 1953282KB, used: 438999KB, 22.47%
15:34:04.281 -> 
15:34:04.281 -> DEBUG [3608]: Arduino Core Version: 2_7_4
[... 384 lines cut ...]
15:37:59.921 -> DEBUG [133007]: Stats: 112 frames/s, frame delay: 0 us
15:37:59.921 -> DEBUG [133007]: Heap: 31544
15:38:00.921 -> DEBUG [134008]: Stats: 111 frames/s, frame delay: 0 us
15:38:01.889 -> DEBUG [135003]: Stats: 111 frames/s, frame delay: 0 us
15:38:02.921 -> DEBUG [136007]: Stats: 112 frames/s, frame delay: 0 us
15:38:03.887 -> DEBUG [137001]: Stats: 111 frames/s, frame delay: 0 us
15:38:04.921 -> DEBUG [138005]: Stats: 112 frames/s, frame delay: 0 us
15:38:05.888 -> DEBUG [139002]: Stats: 111 frames/s, frame delay: 0 us
15:38:06.922 -> DEBUG [140005]: Stats: 112 frames/s, frame delay: 0 us
15:38:07.888 -> DEBUG [141001]: Stats: 111 frames/s, frame delay: 0 us
15:38:08.921 -> DEBUG [142004]: Stats: 112 frames/s, frame delay: 0 us
15:38:09.922 -> DEBUG [143008]: Stats: 112 frames/s, frame delay: 0 us
15:38:09.922 -> DEBUG [143008]: Heap: 31544
15:38:10.922 -> DEBUG [144008]: Stats: 111 frames/s, frame delay: 0 us
15:38:11.890 -> DEBUG [145003]: Stats: 111 frames/s, frame delay: 0 us
15:38:12.921 -> DEBUG [146007]: Stats: 112 frames/s, frame delay: 0 us
15:38:13.888 -> DEBUG [147001]: Stats: 111 frames/s, frame delay: 0 us
15:38:14.889 -> DEBUG [148004]: Stats: 112 frames/s, frame delay: 0 us
15:38:15.921 -> DEBUG [149008]: Stats: 112 frames/s, frame delay: 0 us
15:38:16.888 -> DEBUG [150002]: Stats: 111 frames/s, frame delay: 0 us
15:38:17.921 -> DEBUG [151006]: Stats: 112 frames/s, frame delay: 0 us
15:38:18.887 -> DEBUG [152001]: Stats: 111 frames/s, frame delay: 0 us
15:38:19.920 -> DEBUG [153004]: Stats: 112 frames/s, frame delay: 0 us
15:38:19.920 -> DEBUG [153005]: Heap: 31544
```

bb-Ricardo, 2021-09-08:
This all looks very good. And switching on the mDNS feature (even though this shouldn't change the behavior).

And a `curl -I http://192.168.1.121` results in a timeout?

craigjmidwinter, 2021-09-08:
Yeah, switching on mDNS doesn't seem to change the behaviour. Sometimes it will 200 and come up and be available for a couple minutes and in those cases `curl -I` to the IP will 200 for a little bit, and then it will go unavailable with `curl: (7) Failed to connect to 192.168.1.121 port 80 after 3056 ms: No route to host`. Other times, right out of the gate, it will go to the no route to host error, but there are no errors or anomalies in the log, and the device is still connected to the mqtt broker and responding to commands via mqtt

WarDrake, 2021-09-08:
This sounds like  a server issue, MQTT can repeat it's messages due to QoS so it's not the best comparison for this
Run the Curl command and run a Ping command next to it too
ping -t 192.168.1.121
if ping keeps working while curl doesn't, it means the issue with the webserver itself and we can work from there.

craigjmidwinter, 2021-09-08:
So, so far this thing seems to stay alive as long as I'm pinging it, if I stop after a minute or two I get `Destination Host Unreachable`, but the client is still connected and responding to mqtt messages. Very strange

closed 2022-03-25 by WarDrake