NimmLor/esp8266-fastled-iot-webserver #199: Sleep Mode if LED powered off

thommycgn, 2021-08-23:
The Desk Lamp still consumes 1 W when switched off. I noticed that no sleep mode of the ESP is used at all. Is there a reason for this? Could this perhaps still be implemented?

WarDrake, 2021-08-23:
The ESP and leds all continue to draw power while connected, that's because there are multiple thigns running like a webserver, math calculations, mqtt / alexa alive packages, etc, etc, etc

you could implement a virtual switch like a mosfet to cut the power to the leds and avoid 99% of the power draw, there's a PR for that already made, but no one has had the time to test it, you can try that one if you're interested.

hexenmeister, 2021-08-23:
see here:
https://github.com/NimmLor/esp8266-fastled-iot-webserver/pull/169
I've been using it myself for some time without any problems :)
[image]
[image]
[image]

closed 2022-03-25 by WarDrake