NimmLor/esp8266-fastled-iot-webserver #203: AP not visible after 2nd configuration attempt

JustinHenryLo, 2021-09-04:
Hello, I seem to have encountered a problem trying to connect to the AP. I was able to connect to it the 1st time. But I got disconnected to the AP without correctly inputting the password to the WIFI. After that the AP didnt show up anymore. I tried reuploading the code with a different Hostname, I also tried uploading a sketch with nothing but [wifiManager.resetSettings();] Im not sure if the problem is that it stored the incorrect password thats why the AP isnt showing or if there is something in the code that assumes I correctly inputted so it doesnt bring up the AP again for reconnection? Is there an option where I can just type out the password instead of going through the password portal?

WarDrake, 2021-09-05:
Once you enter any data into the config in the AP, regardless of it is correct or not, a config file is created and the AP is not started anymore, what you need to do is DELETE ALL FLASH DATA not just upload the files again, once you delete the flash data AND uploads the files again, an AP will be generated again.

closed 2021-09-05 by WarDrake

JustinHenryLo, 2021-09-06:
Im sorry I misunderstood my problem, I thought I misinputted the credentials in AP, because the endpoint I was expecting didn't show a webpage with LED controls. But my credentials were correct all along. The Interface was really just showing not found because when I tried using postman to alter the patterns, they just worked fine. Is there a misstep in my configuration?