NimmLor/esp8266-fastled-iot-webserver #204: Very Slow UDP (1 fps)

CastleSeven, 2021-09-23:
I'm trying to use the IoT-Audio-Visualizer project to send spectrum data from my PC over UDP to the Wemos D1 Mini. I've verified using wireshark that plenty of packets are being sent to the LED controllers IP via UDP, so I started looking into the Wemos side of things. 

Using some bracketed timing, it looks like (at least for NanoleafWaves, but likely all the other visualizers as I'm experiencing the same behavior for all visualizer patterns) the parseUdp() call + the ShiftLeds call is taking a full second of time (resulting in ~ 1fps speeds).

[image]

I'll do some more digging on my end to see if 

a) ShiftLeds could be the culprit (not likely given a first pass look)
b) There's a specific portion of parseUdp() that's blocking and if there's a way around it.

CastleSeven, 2021-09-23:
Unfortunately it looks like Udp.ReadBytes() is the culprit:

[image]

CastleSeven, 2021-09-23:
OK, well, I have no idea why but I managed to get it working. After being unable to find a specific reference for "readBytes" in the Arduino UDP Library, I changed readBytes to simply 'read' and the speed is fast enough to nearly keep up now. 

[image]

I'll try to fork and submit a pull request tomorrow, just in case anyone was excited to use their visualizer tonight this may help :)

WarDrake, 2021-09-23:
Hadn't tried it in a while, but you're right, it's taking up to a second to switch frames, please assign me when you make the PR so I can test it

WarDrake, 2021-09-24:
The change to read from readBytes restored the animations to a proper speed.

closed 2021-09-24 by WarDrake