NimmLor/esp8266-fastled-iot-webserver #206: Corrupted MQTT Config Topic

Brackus9, 2021-09-24:
First time raising an issue, so not sure if this is the propper way.

Currently it is possible for the topic to become corrupted during the proces of publishing the config. 
This can cause the connection to be aborted and prevent further communication with the mqtt broker.

The issues is with the creation of the `mqttConfigTopic`

I suspect the problem to arrises due to use of `strlcat` :
https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/1d676ffe696b8821b0f4f1aa87462fafe6cef274/esp8266-fastled-iot-webserver.ino#L1432

In my case this caused the topic to contain some random characters before the actual topic, and ultimately a corrupted packet, causing the broker to drop the connection.

using:
`strlcpy(mqttConfigTopic, cfg.MQTTTopic, sizeof(mqttConfigTopic));`
`strlcat(mqttConfigTopic, "/config",sizeof(mqttConfigTopic));`

as done during the subscription process for example fixes it.

WarDrake, 2021-09-24:
That seems pretty odd...
sounds to me like there might be something else at play, did you happen to have a copy of that corrupt message around?

I'd like to investigate

Brackus9, 2021-09-24:
The message itself looks OK but the topic became corrupted:

Wireshark reports this as topic
`Topic: \004S�?�\003homeassistant/light/twistedlamp/config`

where the extra characters corresponds to the bold bytes in the packet:
offset | byte| acsii
-----| ------------ | -------------
0000 |   31 aa 08 00 2c **04 53 ff 3f fc 03** 68 6f 6d 65 61  |     1...,.S.?..homea
0010 | 73 73 69 73 74 61 6e 74 2f 6c 69 67 68 74 2f 74     |  ssistant/light/t
0020 |  77 69 73 74 65 64 6c 61 6d 70 2f 63 6f 6e 66 69     |  wistedlamp/confi
0030 | 67 7b 22 7e 22 3a 22 68 6f 6d 65 61 73 73 69 73    |   g{"~":"homeassis
0040 |  74 61 6e 74 2f 6c 69 67 68 74 2f 74 77 69 73 74      | tant/light/twist
0050 |  65 64 6c 61 6d 70 22 2c 22 6e 61 6d 65 22 3a 22     |  edlamp","name":"
0060 |  54 77 69 73 74 65 64 20 4c 65 64 20 4c 61 6d 70     |  Twisted Led Lamp
0070 |  22 2c 22 64 65 76 22 3a 7b 22 69 64 73 22 3a 22     |  ","dev":{"ids":"
etc.

The broker reports:
Client Twisted-LED-Lamp disconnected due to malformed packet.

With my limited C knowlegde, I figured it was maybe due to some random bytes in the uninitialized array.

I do also have to note that I am using the latest library versions in the IDE instead of the specific ones, if that matters anything.

closed 2022-03-15 by Brackus9