NimmLor/esp8266-fastled-iot-webserver #212: Cannot connect to server

mrunderhill54, 2021-11-15:
All the uploads appeared to have gone properly. But the connection seems to time out whenever I search http://192.168.4.1/

WarDrake, 2021-11-15:
Delete all the contents of the flash memory, and re upload the data files, that should make the system start the AP in the next boot, but it's important to DELETE EVERYTHING, or you just won't get it working

Elthari0n, 2021-12-12:
hi, i have the same problème, how to delete everythings ?
TKS

birrozza, 2021-12-12:
Hi
In arduino IDE, in the tools menu, in  "Erase Flash" you must select  "All Flash Contents" item.
Then reflash the sketch.
After, you must re-upload the data file with  "ESP8266 Sketch Data Upload" in tools menu of Arduino IDE

Elthari0n, 2021-12-12:
I did not understand everything, I correctly put "all flash contents" and upload again to the card.
I did not understand how to reload the data file? download it on arduino or on the card? sorry this is my 1st use ...

birrozza, 2021-12-12:
You probably don't have the function "ESP8266 Sketch Data Upload" in the arduino ide menu.
I think this video can help you: https://youtu.be/vgmZpv47Dj0

Elthari0n, 2021-12-12:
j'ai bien cette fonction et je l'ai fait après le flash mais le problème ne change pas

Elthari0n, 2021-12-12:
[image]
[image]
is this the right module?

birrozza, 2021-12-12:
I have the same board/module and for me it work well.
If it is the first time you start the sketch on the module, then you have to configure the wifi connection. In this case the ESP8266 starts as an access point. Then you connect to its wifi network (in my case it is "LEDs - F385) with your smartphone / pc and with the browser go to the address http://192.168.1.4 and configure the wifi connection via the wifimanager. If all ok, the module auto restart and you can find it in your network with the default host name "LEDs"

Elthari0n, 2021-12-12:
I finally managed to connect to the card, I was able to add it to my wifi, and on my box the IP address is 192.168.1.137, when I put it in the search engine I come across the page with color changes, but it doesn't seem syncronized ...
do you have to install an application or is everything done from the web browser?

birrozza, 2021-12-12:
The module is controlled via the web page.
I am not aware of any applications.

Elthari0n, 2021-12-12:
Thank you for your answers.
my problem is that currently I can't give orders from the website and I don't understand why. I am on the same wifi as the card.

I have only welded one triangle so far, is it possible that the problem comes from there?

birrozza, 2021-12-12:
In the sketch you must set the typology of your leds. In my  case it is a strip leds of 20 leds and the _#define LED_DEVICE_TYPE_ is set to 0.
You must be sure of the type of your LEDs

Elthari0n, 2021-12-13:
Hello,

I have checked the type of led and set function 4 for nanoleaf, with 1 tirange and 12 leds per triangle but the problem remains the same, (ditto when I just set to 0 like you) as if the information was not not received by the card.

Elthari0n, 2021-12-13:
I move forward little by little!
my problem was that the data_pin was on 3, instead of 4.
now i have 6 triangles which light up out of 15, the last led is green as if there was an error ... has anyone already had this?