NimmLor/esp8266-fastled-iot-webserver #213: Bootloops after WiFiManager setup

hntirgeam, 2021-11-17:
Reboots immediately after wifiManager.autoConnect(nameChar) (when settings already stored)

Board: NodeMCU

`Basic.ini` from WiFiManager repo works fine and connects to my wifi without any problems. 

Tried: 
- Changing CPU freq from 80 to 180
- Adding `setConnectTimeout(60)` before `autoConnect();`
- Adding `#define FASTLED_ESP8266_NODEMCU_PIN_ORDER`
- Full flash erase
- Hardcoding wifi log/pass in `wifiManager.autoConnect()`

First setup log (after full flash erase)

```
DEBUG [178]: Loading config
DEBUG [179]: System Information:
DEBUG [179]: Version: 4.5 (2020-02-14)
DEBUG [180]: Heap: 38536
DEBUG [180]: SDK: 2.2.2-dev(38a443e)
DEBUG [181]: Boot Vers: 31
DEBUG [184]: CPU Speed: 80 MHz
DEBUG [186]: Chip ID: 14765809
DEBUG [189]: Flash ID: 1458392
DEBUG [192]: Flash Size: 4194304KB
DEBUG [195]: Vcc: 65535
DEBUG [197]: MAC address: E8:DB:84:E1:4E:F1

*WM: [1] getCoreVersion():          3.0.2
*WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             36336
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  LEDs - 4EF1
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             LEDs - 4EF1
*WM: [1] password:         
*WM: [1] ssid_len:         11
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 1 ms
*WM: [2] Config Portal Running, non blocking/processing 
INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP 'LEDs - 4EF1' to configure your wireless connection
DEBUG [347]: SPIFFS contents:
DEBUG [355]: FS File: /css/bootstrap.min.css, size: 121093
DEBUG [355]: FS File: /css/dark-mode.css, size: 1350
DEBUG [363]: FS File: /css/minicolors.min.css, size: 96597
DEBUG [364]: FS File: /css/styles.css, size: 1392
DEBUG [369]: FS File: /favicon.ico, size: 4286
DEBUG [373]: FS File: /fonts/glyphicons.woff2, size: 18028
DEBUG [378]: FS File: /images/atom196.png, size: 3889
[... 193 lines cut ...]
0x4020a94c: ESP8266WiFiSTAClass::status() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 527
0x4021a7b7: __delay(unsigned long) Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 53
0x4020a98b: ESP8266WiFiSTAClass::waitForConnectResult(unsigned long) Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/PolledTimeout.h line 260
0x4020eca0: WiFiManager::waitForConnectResult(unsigned short) at H:\Software\Arduino IDE New\Arduino\libraries\WiFiManager-2.0.3-alpha\WiFiManager.cpp line 912
0x4020ee64: WiFiManager::waitForConnectResult() at H:\Software\Arduino IDE New\Arduino\libraries\WiFiManager-2.0.3-alpha\WiFiManager.cpp line 901
0x4020aa14: ESP8266WiFiSTAClass::begin() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 237
0x4020aa01: ESP8266WiFiSTAClass::begin() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 232
0x4020aa1c: ESP8266WiFiSTAClass::begin() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 237
0x4020e30a: WiFiManager::wifiConnectDefault() at H:\Software\Arduino IDE New\Arduino\libraries\WiFiManager-2.0.3-alpha\WiFiManager.cpp line 842
0x4020c2c4: String::operator!=(char const*) const Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 155
0x4020f66a: WiFiManager::connectWifi(String, String) at H:\Software\Arduino IDE New\Arduino\libraries\WiFiManager-2.0.3-alpha\WiFiManager.cpp line 784
0x40218edc: String::copy(char const*, unsigned int) Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\WString.cpp line 225
0x40218f15: String::String(char const*) Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\WString.cpp line 41
0x40211f7a: WiFiManager::autoConnect(char const*, char const*) at H:\Software\Arduino IDE New\Arduino\libraries\WiFiManager-2.0.3-alpha\WiFiManager.cpp line 297
0x4020f8b5: WiFiManager::debugPlatformInfo() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x40204fa4: setup() at  Desktop\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 770
0x40204f43: setup() at  Desktop\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 755
0x40207b34: handleReboot() at  Desktop\esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1600
0x40223fa4: std::_Function_handler ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 268
0x40223f90: std::_Function_handler ::_M_invoke(std::_Any_data const&) arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 289
0x4021a1c4: loop_wrapper() Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 198

```

Any ideas? Thanks.

bb-Ricardo, 2021-11-17:
Hi, did you use the tested library versions?: https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md#dependencies

hntirgeam, 2021-11-26:
Yes, I used the recommended library.