NimmLor/esp8266-fastled-iot-webserver #214: Home assistant mqtt light not discovery but mqtt explore show message

1975felix, 2021-12-09:
In mqtt explore i see the massage but in home assistant get this error:

Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'name': 'Hexaleafs', 'device': {'identifiers': 'Hexaleaf_lamp', 'manufacturer': 'WD DIY', 'model': '0.1', 'name': 'DIY Nanoleafs'}, 'state_topic': 'homeassistant/light/hexaleaf', 'command_topic': 'homeassistant/light/hexaleaf/set', 'brightness': True, 'rgb': True, 'effect': True, 'unique_id': 'hexaleaf_lamp', 'schema': 'json', 'effect_list': ['Pride', 'Color Waves', 'Horizontal Rainbow', 'Solid Rainbow', 'Confetti', 'Sinelon', 'Beat', 'Juggle', 'Fire', 'Water', 'Strobe', 'Rainbow Strobe', 'Smooth Rainbow Strobe', 'Rainbow Twinkles', 'Snow Twinkles', 'Cloud Twinkles', 'Incandescent Twinkles', 'Retro C9 Twinkles', 'Red & White Twinkles', 'Blue & White Twinkles', 'Red, Green & White Twinkles', 'Fairy Light Twinkles', 'Snow 2 Twinkles', 'Holly Twinkles', 'Ice Twinkles', None], 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 198, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 38, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string value is None @ data['effect_list'][25]

bb-Ricardo, 2021-12-09:
Hi, it seems like the last element in the effect list is None. 

Can you compile the code with DEBUG set to 1 and poat the serial output here? That would help a lot.

Thank you.

1975felix, 2021-12-09:
Thant for answare, i'm look for led Debug set but no line in sketch, becouse i have a old version.
Now i update to new one enable led debug but now wemos reboot continuously after few second...

WarDrake, 2021-12-09:
Old versions of the script would not allow a full MQTT configuration message to be generated while visualizations were enabled.
this is solved in newer versions, if you wish to continue using the old one, disable audio visualizations.

WarDrake, 2022-03-07:
That error is not normal, get a log of the nanoleafs side as they connect and post that, create a new issue for it too please.