NimmLor/esp8266-fastled-iot-webserver #215: GradientPalettes.h: No such file or directory

Elthari0n, 2021-12-12:
hello, sorry for my bad english (google translate)
I followed the instructions and when downloading the code on the card, the following error is displayed. Can you help me please 

Arduino : 1.8.16 (Windows 10), Carte : "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

esp8266-fastled-iot-webserver:34:30: fatal error: GradientPalettes.h: No such file or directory

compilation terminated.

exit status 1

GradientPalettes.h: No such file or directory

Elthari0n, 2021-12-12:
this issue is fixed, the folder name was not the correct esp8266-fastled-iot-webserver

another problem, to make the connection to the esp8266, when I type 192.168.4.1 in the browser, no answer ... what to do please?

iwangrizzlee, 2022-09-04:
@Elthari0n 
I have the same problem. tell me how to solve it?

festivalist, 2022-10-26:
> 

I have the same problem, did you figure it out?