NimmLor/esp8266-fastled-iot-webserver #217: WebUI wont connect

Roycinger, 2021-12-20:
Hi there,

I just reflashed my Wemos D1 with the sketch and noticed that the WebUI does not fully load.
A similar issue is reported in this issue here: https://github.com/NimmLor/esp8266-fastled-iot-webserver/issues/190

After reading this, I cleared my EEPROM (with the Arduino sketch from Examples -> EEPROM), chose Tool -> Erase Flash: "All flash contents", this is what I assume is a "clean" install.

After that, I connected to the AP that's provided by the D1 and connected it to my Wifi (which connected successfully):
[image]

Also, I uploaded the Sketch data via Tools -> ESP8266 Sketch Data Upload.

After all that, I tried accessing the WebUI and get the same issue as in the issue linked above.
It says "Connecting, please wait" at the bottom and wont load any further:
[image]

Any help to resolve this is greatly appreciated

WarDrake, 2022-01-13:
Press F12 and select console in the new window, after that reload the page, wait 30 seconds and post the output of the console over here, that'll help us diagnose why it's not connecting.

Ceda, 2022-01-31:
@WarDrake Same issue...
[image]

WarDrake, 2022-01-31:
Hmm... I suppose your power source could be delivering too high  a voltage, amperage is not an issue on these ESP boards, check it really is delivering 5v power under load I think the most the ESP can take before it fails is 8 or so...
how are you powering the leds? from the ESP or from the powersource directly?

Ceda, 2022-01-31:
> Hmm... I suppose your power source could be delivering too high a voltage, amperage is not an issue on these ESP boards, check it really is delivering 5v power under load I think the most the ESP [...]

On the contrary, the voltage in the USB programmer was probably low. Now is working correct.

WarDrake, 2022-01-31:
Ah, that makes sense, it's the pins on the ESP-01 when all pins are connected there's a short between 2 of them to set into programming mode, that's what prevents the webserver from working properly, you might be able to set the programmer to read mode only if not, then run it outside the programmer