NimmLor/esp8266-fastled-iot-webserver #223: Calls for FS.h in script, but the file doesnt exist and isnt provided

emersonsc, 2022-01-11:
/*
   ESP8266 FastLED WebServer: https://github.com/jasoncoon/esp8266-fastled-webserver
   Copyright (C) 2015-2018 Jason Coon

   This program is free software: you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation, either version 3 of the License, or
   (at your option) any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#define FASTLED_INTERRUPT_RETRY_COUNT 1
//#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
FASTLED_USING_NAMESPACE
**#include <FS.h>**
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
}
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#elif defined(ESP32)
#include <WiFi.h>
#include <SPIFFS.h>
#endif
#include <EEPROM.h>
#include "GradientPalettes.h"
#include "Field.h"

Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\emers\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\emers\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\emers\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10819 -build-path C:\Users\emers\AppData\Local\Temp\arduino_build_711079 -warnings=none -build-cache C:\Users\emers\AppData\Local\Temp\arduino_cache_571088 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose A:\Emersons Stuff\NanoShit\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\emers\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\emers\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\emers\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10819 -build-path C:\Users\emers\AppData\Local\Temp\arduino_build_711079 -warnings=none -build-cache C:\Users\emers\AppData\Local\Temp\arduino_cache_571088 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose A:\Emersons Stuff\NanoShit\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino

Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\emers\\AppData\\Local\\Temp\\arduino_build_711079\\sketch\\esp8266-fastled-iot-webserver.ino.cpp" -o nul

Alternatives for FastLED.h: [FastLED@3.5.0]

ResolveLibrary(FastLED.h)

  -> candidates: [FastLED@3.5.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\emers\\OneDrive\\Documents\\Arduino\\libraries\\FastLED\\src" "C:\\Users\\emers\\AppData\\Local\\Temp\\arduino_build_711079\\sketch\\esp8266-fastled-iot-webserver.ino.cpp" -o nul

Alternatives for FS.h: []esp8266-fastled-iot-webserver:22:10: fatal error: FS.h: No such file or directory

ResolveLibrary(FS.h)

          ^~~~~~

  -> candidates: []compilation terminated.

Using library FastLED at version 3.5.0 in folder: C:\Users\emers\OneDrive\Documents\Arduino\libraries\FastLED 

exit status 1

FS.h: No such file or directory

bb-Ricardo, 2022-01-11:
Hi,

did you set `ENABLE_ALEXA_SUPPORT` to `1`?

WarDrake, 2022-01-12:
for MQTT it looks like you don't have the PubSubClient library installed on your arduino IDE, try insalling that and see if it still throws an error.

emersonsc, 2022-01-12:
> for MQTT it looks like you don't have the PubSubClient library installed on your arduino IDE, try insalling that and see if it still throws an error.

Its all there:

#include <ArduinoJson.h>
#include <PubSubClient.h>
#include <ArduinoOTA.h>
#include <Espalexa.h>
#include <EspalexaDevice.h>
#include <strings_en.h>
#include <WiFiManager.h>
#include <FastLED.h>

emersonsc, 2022-01-12:
> Hi,
> 
> did you set `ENABLE_ALEXA_SUPPORT` to `1`?

It didnt have the 1, so i tried it with the 1, and still no go. I get this:

Alexa pairing mode
Procedure:
The webserver will reboot and the UI won't be available.
Now. Say to Alexa: 'Alexa, discover devices'.

Alexa should tell you that it found a new device, if it did reset the esp8266 to return to the normal mode.
Exit pairing mode: Reboot

But my Alexa never finds it.

WarDrake, 2022-01-12:
AH, my bad, I see what the Issue is, Lines 198 to 208 define specific patterns for Alexa, you need to uncomment those as well, and then the alexa errors will go away, after that whatever other errors you might be facing should be back or the software will compile properly

From the line numbers I'm seeing in your errors compared to the ones on the latest version of the script, you need to update to the latest version as well as you have approximately 10 lines missing. which might actually be those 198 to 208 lines.

emersonsc, 2022-01-12:
> AH, my bad, I see what the Issue is, Lines 198 to 208 define specific patterns for Alexa, you need to uncomment those as well, and then the alexa errors will go away, after that whatever other [...]

Is it ok to install recent versions of the scripts then? The install lists outdated scripts, but I wanted to make sure I installed all the versions listed in the installation.md

<html>
<body>
<h2 dir="auto" style="box-sizing: border-box; margin-top: 24px; margin-bottom: 16px; font-size: 1.5em; font-weight: 600; line-height: 1.25; padding-bottom: 0.3em; border-bottom: 1px solid var(--color-border-muted); color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Dependencies</h2><ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="box-sizing: border-box;">esp8266 and esp32 libs have to be installed via board manager</li><li style="box-sizing: border-box; margin-top: 0.25em;">alle other libs need to be installed via library manager</li></ul>

Name | Version | Type | Comment
-- | -- | -- | --
esp8266 | 2.7.4 | required | esp8266 Core Library if ESP8266 board is used
esp32 | 1.0.4 | required | esp32 Core Library if ESP32 board is used
FastLED | 3.4.0 | required | LED animation library
WiFiManager | 2.0.3-alpha | required | easy WiFi setup integration
Espalexa | 2.5.0 | optional | library for Alexa integration
PubSubClient | 2.8.0 | optional | for MQTT/Homeassistant support
ArduinoJson | 6.17.2 | optional | for MQTT/Homeassistant support

</body>
</html>

WarDrake, 2022-01-12:
I was talking about the main script, not it's dependencies, as for the dependencies, newer versions should be ok for most of them except FastLED as the newer versions cause an issue with one led always being active and/or flashing a different color than expected.

emersonsc, 2022-01-12:
> I was talking about the main script, not it's dependencies, as for the dependencies, newer versions should be ok for most of them except FastLED as the newer versions cause an issue with one led always being active and/or flashing a different color than expected.

I got it from the latest release from here:
https://github.com/NimmLor/esp8266-fastled-iot-webserver/releases

WarDrake, 2022-03-25:
Can you check if it works with the new version please?

emersonsc, 2022-03-26:
> Can you check if it works with the new version please?

Yeah I can do that. I set the project aside for a bit. While im updating that, should I update any of the dependencies as well? Should I make sure FastLED stays at 3.4 or is that safe to update as well?

WarDrake, 2022-03-26:
It's safe to update FastLED but you'll get flickering on the first led with newer versions, if you want perfect control over the leds, stay on the older versions.
The rest of the dependencies have no issues.