NimmLor/esp8266-fastled-iot-webserver #224: No hardware SPI pins defined

emersonsc, 2022-01-11:
In file included from A:\Emersons Stuff\NanoShit\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:13:0:
C:\Users\emers\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
In file included from C:\Users\emers\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,
                 from A:\Emersons Stuff\NanoShit\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver-4.5\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:13:
C:\Users\emers\OneDrive\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^

bb-Ricardo, 2022-01-11:
Hi, I have this problem as well. I compiles and uploads fine. at least on my Wemos 1D.

emersonsc, 2022-01-11:
> Hi, I have this problem as well. I compiles and uploads fine. at least on my Wemos 1D.

Does it work with this project overall?

WarDrake, 2022-01-12:
No Hardware SPI pins is not an issue, it's just a message and it's present in all software using FASTLed libraries, as they don't use the serial port interface for their messages.

closed 2022-01-12 by WarDrake