NimmLor/esp8266-fastled-iot-webserver #229: MQTT not working

xbavnu, 2022-03-04:
I am having problems setting up MQTT with my light panels. I enabled MQTT support in the configuration before flashing the software and it is also enabled and configured in the settings in the web interface. I can control the lighs via HTTP requests but I am unable to use them with alexa or home assistant. 

The MQTT log is flooded with these messages:
1646414788: Socket error on client Nanoleaf, disconnecting.
1646414798: New connection from 192.168.178.25 on port 1883.
1646414798: New client connected from 192.168.178.25 as Nanoleaf (p2, c1, k10, u'mqtt-user').

Here is a picture of my MQTT configuration on the web interface:
[image]

What am I missing or what could be the problem here?

xbavnu, 2022-03-05:
I also enabled the debugging messages on the Wemos D1 mini and this is a snippet with all the messages that are repeating all the time:

> Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
DEBUG [163007]: Stats: 105 frames/s, frame delay: 0 us
DEBUG [163008]: Connecting to MQTT...connected
DEBUG [163025]: Subscribing to MQTT Topics
DEBUG [163030]: Configuration Publishing Begun
DEBUG [163055]: Configuration Sent
DEBUG [163056]: Configuration Publishing Finished
DEBUG [163072]: Sending MQTT package: {"state":"ON","brightness":200,"effect":"Solid Color","autoplay":0,"speed":2,"hue":68,"saturation":122}
DEBUG [163073]: Sending Initial Status
DEBUG [163077]: Heap: 23656
DEBUG [164007]: Stats: 99 frames/s, frame delay: 0 us
DEBUG [165007]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [166008]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [167008]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [168008]: Stats: 106 frames/s, frame delay: 0 us
AlexaApiCall
ok
l1442614144
1442614144
DEBUG [169009]: Stats: 105 frames/s, frame delay: 0 us
DEBUG [170009]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [171009]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [172000]: Stats: 105 frames/s, frame delay: 0 us
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
Got UDP!
DEBUG [173007]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [173008]: Connecting to MQTT...connected
DEBUG [173015]: Subscribing to MQTT Topics
DEBUG [173019]: Configuration Publishing Begun
DEBUG [173045]: Configuration Sent
DEBUG [173045]: Configuration Publishing Finished
DEBUG [173050]: Sending MQTT package: {"state":"ON","brightness":200,"effect":"Solid Color","autoplay":0,"speed":2,"hue":68,"saturation":122}
DEBUG [173054]: Sending Initial Status
DEBUG [173057]: Heap: 23656
DEBUG [174006]: Stats: 101 frames/s, frame delay: 0 us
DEBUG [175006]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [176006]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [177006]: Stats: 106 frames/s, frame delay: 0 us
DEBUG [178005]: Stats: 106 frames/s, frame delay: 0 us

WarDrake, 2022-03-07:
Everything seems good with your setup and log, it might just be the autodiscovery feature on HA that might be off, try enabling it

https://www.home-assistant.io/docs/mqtt/discovery/

xbavnu, 2022-03-10:
> Everything seems good with your setup and log, it might just be the autodiscovery feature on HA that might be off, try enabling it
> 
> https://www.home-assistant.io/docs/mqtt/discovery/

Thank you for the response. Auto discovery is enabled and works just fine with other MQTT devices running tasmota.

In the home assistant MQTT log I still have these messages popping up every couple seconds:
1646414788: Socket error on client Nanoleaf, disconnecting.
1646414798: New connection from 192.168.178.25 on port 1883.
1646414798: New client connected from 192.168.178.25 as Nanoleaf (p2, c1, k10, u'mqtt-user').

So it seems like the ESP is connecting and disconnecting all the time.

Brackus9, 2022-03-13:
I had a similar issue (see issue #206 ). Using strlcpy instead of strlcat, on line 1432 fixed the issue where the esp would continuously try to reconnect.

xbavnu, 2022-03-14:
@Brackus9 

Thank you very much! It seems like this fixed all the problems. I can now use both, Alexa and MQTT to control the lights.

WarDrake, 2022-03-14:
@Brackus9 would you mind submitting a PR with that change please?

closed 2022-03-15 by xbavnu