NimmLor/esp8266-fastled-iot-webserver #235: ESP8266 NodeMcu can't figure out proper data pin

Tpow99, 2022-06-25:
Hey! I am working on a nodemcu device and can't seem to get any of the pins to work as the data pin to the lights. When I have the device configured and wifi connections and local host both working, only the first three lights seem to power on at all, they flash random colors and the data pin cannot even turn them off, any thoughts?

WarDrake, 2022-06-26:
NodeMCU requires a couple small changes, you can go to my form and use the NodeMCU branch there

Tpow99, 2022-06-27:
@WarDrake thanks! I will give that a try this evening

Tpow99, 2022-06-28:
@WarDrake what pin would be pin 0? In your example, or would I name my own pin D4, D2, etc?

Tpow99, 2022-06-28:
Ah, a clean flash and a good connection and all 300 lights came to life! They are twinkling instead of solid, but you finally got me to a point I can work with! Thanks you @WarDrake

Tpow99, 2022-07-09:
@WarDrake sorry to sort of spam you here, still having issues after taking a break for a while, I am attempting to use DATA_PIN: D2 and still seem to have no luck at all, any advice?

WarDrake, 2022-07-09:
Use raw numbers, not Digital numbers. there's a line at the top of that branch to use raw numbers  on line #18

#define FASTLED_ESP8266_RAW_PIN_ORDER       //To use raw NodeMCU pin numbers

[image]